espressif/arduino-esp32 #3055: Heap corruption with mbedtls mbedtls_pk_decrypt

nikojarvinen, 2019-08-01:
I'm trying to build a simple script that would encrypt and then decrypt a string using private/public keys. The encryption part seems to work ok but the decryption part crashes esp32. I've been trying anything I can think of but can't get it to work for some reason. 

I'm on windows 10 and running ubuntu on WSL. ESP-IDF is on 4.0-dev-1287

The full code to replicate this is below:
The line where it seems to be crashing is:
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

#include <string.h>
#include "mbedtls/pk.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"

int ret = 0;

void app_main()
{

    mbedtls_entropy_context entropy;
    mbedtls_entropy_init( &entropy );

    mbedtls_ctr_drbg_context ctr_drbg;
    char *personalization = "my_app_specific_string";

    mbedtls_ctr_drbg_init( &ctr_drbg );

    ret = mbedtls_ctr_drbg_seed( &ctr_drbg , mbedtls_entropy_func, &entropy, (const unsigned char *) personalization, strlen( personalization ) );
    if( ret != 0 )    {
        // ERROR HANDLING CODE FOR YOUR APP
    }

//begin encryption part

    mbedtls_pk_context pk;
    mbedtls_pk_init( &pk );

static const char publicKey[] = "-----BEGIN PUBLIC KEY-----\n"\
"MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAw/onKUJSaPh3mPVhUonM\n"\
"/H3pv3t8aIxeN6BMV9l8vwjwrcyWSso6XrG8MOVLGpcyrb31D/J0ayYI2btXx5hk\n"\
"ouoKcWHgQUM0/ChV98jpoPB7436vIxXc3HVSrfLKxjN9Ja3RZAsvn26OFJOthZ/M\n"\
"82Od5w5YHDZnzs9p4RiuUoF5cX/HP9nT2Kc4tudx80HQKkOgWyyGZtamBUT2AkSf\n"\
"fLep67TeO+ErKrsS6wDb8jwkS7FB1R3BwmeKXyyIp31jAIn0TGLYjKSepJ6gHWlo\n"\
"Q45SHuzl76r5sgSCaPvVojU9Wu7bghxPefAmlBSLHNtfzZK+q/wCm47xuVGbiYZ0\n"\
"SwIDAQAB\n"\
"-----END PUBLIC KEY-----\n";

        static const char to_encrypt[] =  "testing stridasoijqwjeugqwdg";
        uint8_t to_encrypt_len = sizeof(to_encrypt);
        printf("length: %d \n", to_encrypt_len);
        printf("length2: %d \n", sizeof(publicKey));
    /*
     * Read the RSA public key
     */

    if( ( ret = mbedtls_pk_parse_public_key( &pk, (const unsigned char *) publicKey, sizeof(publicKey) ) ) != 0 )    {
        printf( " failed\n  ! mbedtls_pk_parse_public_key returned -0x%04x\n", -ret );
    }
    
    
   //unsigned char buf[MBEDTLS_MPI_MAX_SIZE];
   unsigned char buf[270];
    size_t olen = 0;

    /*
     * Calculate the RSA encryption of the data.
[... 241 lines cut ...]
0x400d757d: mbedtls_calloc at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/platform.c:66

0x400dc87e: mbedtls_mpi_grow at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:132

0x400dd997: mbedtls_mpi_exp_mod at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1897 (discriminator 16)

0x400d7781: rsa_prepare_blinding at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:798

0x400d7dca: mbedtls_rsa_private at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:926

0x400d8495: mbedtls_rsa_rsaes_pkcs1_v15_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:1597 (discriminator 1)

0x400d861d: mbedtls_rsa_pkcs1_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:1735

0x400d651a: rsa_decrypt_wrap at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:132

0x400d63ef: mbedtls_pk_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:439

0x400d3ffc: app_main at /home/testing/esp/hello_world/build/../main/hello_world_main.c:137

0x400d0fca: main_task at /home/testing/esp/esp-idf/components/esp32/cpu_start.c:549

0x40086fd9: vPortTaskWrapper at /home/testing/esp/esp-idf/components/freertos/port.c:143

Rebooting...```

atanisoft, 2019-08-01:
this should be filed over on https://github.com/espressif/esp-idf/ as this does not involve the arduino-esp32 code.

nikojarvinen, 2019-08-01:
whops sorry. I'll move it.

closed 2019-08-01 by nikojarvinen