espressif/arduino-esp32 #3056: A bug when connecting to not-available endpoints

gilmaimon, 2019-08-01:
My hardware:
Node-MCU ESP32 was used, but I was able to reproduce with any esp32 module I had around me.

I am trying to connect to an endpoint which I know does not exist, however the method `WiFiClient.connect` returns `true` (should be `false`). Then, any calls for `send` will crash the esp32.

The code I'm using to reproduce is exactly the same as the sketch `WiFiClientBasic`:
```c++
/*
 *  This sketch sends a message to a TCP server
 *
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti WiFiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network
    WiFiMulti.addAP("apname", "pass");

    Serial.println();
    Serial.println();
    Serial.print("Waiting for WiFi... ");

    while(WiFiMulti.run() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    delay(500);
}

void loop()
{
    const uint16_t port = 80;
    const char * host = "192.168.1.1"; // ip or dns

    Serial.print("Connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;

    if (!client.connect(host, port)) {
        Serial.println("Connection failed.");
        Serial.println("Waiting 5 seconds before retrying...");
        delay(5000);
        return;
    }

    // This will send a request to the server
    client.print("Send this data to the server");

    //read back one line from the server
    String line = client.readStringUntil('\r');
    client.println(line);

    Serial.println("Closing connection.");
    client.stop();

    Serial.println("Waiting 5 seconds before restarting...");
    delay(5000);
}
```

Notice that `client.connect(host, port)` returns `true`, even tho the endpoint doesn't exist!

The crash stacktrace (decoded):
```
PC: 0x400d1b14: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi\src\WiFiClient.cpp line 107
EXCVADDR: 0x00000008

Decoding stack results
0x400d1b14: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi\src\WiFiClient.cpp line 107
0x400d1be5: WiFiClient::read(unsigned char*, unsigned int) at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi\src\WiFiClient.cpp line 407
0x4013db16: WiFiClient::read() at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi\src\WiFiClient.cpp line 318
0x400d3251: Stream::timedRead() at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\Stream.cpp line 36
0x400d3285: Stream::readStringUntil(char) at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\Stream.cpp line 287
0x400d187f: loop() at C:\Users\<username>\AppData\Local\Temp\arduino_modified_sketch_306897/WiFiClientBasic.ino line 59
0x400d3d15: loopTask(void*) at C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 19
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

boarchuz, 2019-08-02:
A fix was merged recently that should address this. Can you update and check if it's still happening?

gilmaimon, 2019-08-02:
I've tested with the latest release I could find on the Boards Manager - 1.0.3-rc1 and the issue still occurs

Thanks for the quick response, 
Gil.

a-c-sreedhar-reddy, 2019-08-02:
Dont test with the latest release , test with the latest master commit.

boarchuz, 2019-08-02:
Yeah I just tested with latest master and confirmed it is fixed now (and reproduced the error on 1.0.3-rc1).
You'll have to wait for the next release for it to be available in Boards Manager.

gilmaimon, 2019-08-02:
Ok, didn't know that.

Thanks guys, I'm waiting for the next stable release.

Gil.

closed 2019-08-02 by gilmaimon