espressif/arduino-esp32 #3057: Esp32  flash read err, 1000 ets_main.c 371 rebooting

mmashraf, 2019-08-02:
I buy new ESP32 from Ali express and after first program of blink , it is continuously showing the sw_reset issue on serial port. i also try python esptool.py erase_flash but it does not work and also following error. I explore all forums but get no resolution and positive response to fix it. Please help me to sort it.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

lbernstone, 2019-08-02:
Sounds like your code upload is failing
See https://github.com/espressif/arduino-esp32/issues/3036

TheMrPhantom, 2019-08-09:
I had this problem after connecting 2 leds on different grounds, so maybe it's a wiring problem?
After connecting the leds to the same ground it worked properly again

Mbonea-Mjema, 2019-08-12:
I have the same problem, I am even not able to erase the chip....

closed 2019-10-25 by stale[bot]

hdpklm, 2021-08-21:
i have dthe same problem and fix it with changing the settings of programmer in Adruino IDE

i think is find start program from an address not exist in nand flash

[image]