espressif/arduino-esp32 #3058: Client Server communication between esp32 and esp8266

hazee007, 2019-08-02:
Hello guys 
am trying to use to use esp32 as a server and two esp32 as a clients, 
i have a code for connecting two esp8266 together working but when i use the esp8266 code on esp32 after changing the #include <ESP8266WiFi.h> to #include <WiFi.h> it got uploaded but i am getting a different access point from the one defined in the code.

code for Server
`//-- Libraries Included --------------------------------------------------------------
  #include <WiFi.h>

//------------------------------------------------------------------------------------
  // Define I/O Pins
  #define     LED0        2         // WIFI Module LED

//------------------------------------------------------------------------------------
  // Authentication Variables
  char*       ssid;              // SERVER WIFI NAME
  char*       password;          // SERVER PASSWORD

//------------------------------------------------------------------------------------
  // WiFi settings
  #define     MAXSC     6           // MAXIMUM NUMBER OF CLIENTS

  IPAddress APlocal_IP(192, 168, 4, 1);
  IPAddress APgateway(192, 168, 4, 1);
  IPAddress APsubnet(255, 255, 255, 0);

  unsigned int TCPPort = 2390;
 
  WiFiServer  TCP_SERVER(TCPPort);      // THE SERVER AND THE PORT NUMBER
  WiFiClient  TCP_Client[MAXSC];        // THE SERVER CLIENTS Maximum number
//------------------------------------------------------------------------------------
  // Some Variables
  char result[10];

void setup(){
  
  // Setting the serial port
  Serial.begin(115200);           // Computer Communication
    
  // Setting the mode of the pins
  pinMode(LED0, OUTPUT);          // WIFI OnBoard LED Light
    
  // setting up a Wifi AccessPoint
  SetWifi("DataTransfer","");
}

//====================================================================================
 
void loop(){
  
  HandleClients(); 

}

//====================================================================================
 
void SetWifi(char* NameESP, char* Password123){
  // Stop any previous WIFI
  WiFi.disconnect();

  // Setting The Wifi Mode
  WiFi.mode(WIFI_AP_STA);
  Serial.println("WIFI Mode : AccessPoint");
  
  // Setting the AccessPoint name & password
  ssid      = NameESP;
  password  = Password123;
   
  // Starting the access point
  WiFi.softAPConfig(APlocal_IP, APgateway, APsubnet);                 // softAPConfig (local_ip, gateway, subnet)
  WiFi.softAP(ssid, password, 1, 0, MAXSC);                           // WiFi.softAP(ssid, password, channel, hidden, max_connection)     
[... 193 lines cut ...]
    Serial.print  ("Device IP Address : ");
    Serial.println(WiFi.localIP());
 
  // conecting as a client -------------------------------------
    Tell_Server_we_are_there();
  }
}

//====================================================================================

void Tell_Server_we_are_there(){
  // first make sure you got disconnected
  TCP_Client.stop();

  // if sucessfully connected send connection message
  if(TCP_Client.connect(TCP_Server, TCPPort)){
    Serial.println    ("<" + Devicename + "-CONNECTED>");
    TCP_Client.println ("<" + Devicename + "-CONNECTED>");
  }
  TCP_Client.setNoDelay(1);                                     // allow fast communication?
}

//====================================================================================`

PS Both code are working perfect between 2 esp8266 .

hazee007, 2019-08-02:
I tried using the `insert code ` 
dont know why it give the code like that.

everslick, 2019-08-02:
it is three ticks \`\`\` before and after the snipped!

hazee007, 2019-08-03:
> it is three ticks ``` before and after the snipped!

Thanks Man... Still Looking for solutions tho

closed 2019-10-16 by stale[bot]