espressif/arduino-esp32 #3059: Compiling on CentOs 6 (Missing glibc-2.14 and GLIBCXX_3.4.15)

kzka90, 2019-08-02:
I'm trying to compile a .INO file within my server, I have a 64-bit CentOs 6

At first, it was telling me I needed a higher glibc version than the one I had, so I added the one it was asking to my server, I didn't installed it as the main but as a parallel compiler, everytime I want to use it, I'm doing this:

```
LD_LIBRARY_PATH=/opt/glibc-2.14/lib
export LD_LIBRARY_PATH
```

After setting this glibc version, I tried again to run the compiling code:

`./arduino-cli compile --fqbn esp32:esp32:esp32 test`

But now I'm encountering the following:

```
Build options changed, rebuilding all
/root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.15' not found (required by /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus)
Error: build failed: exit status 1
Error during build
```

Seems that I need a higher version of the libstdc++.so.6 library, it's asking for: **GLIBCXX_3.4.15**

I've searched everywhere on how to install this in CentOs 6 (without compromising my server) and have found nothing.

I have two options:

1. Find a way to use **GLIBCXX_3.4.15** and hope there won't be other missing parts to compile.
2. Find a way to compile the desire code with the libraries my server already have.

Any ideas what to do?

kuoyaoming93, 2019-08-17:
@kzka90 

Hi Carlos,

I had the similar issue working with Centos7, compiling using another IDEs and another platform.
But I think that the problem is the same. GCC version is too old in yum repo. You should compile GCC from source.

I have a script at work, on Tuesday I could send you.

kuoyaoming93, 2019-08-30:
@kzka90 

Use this script to update your gcc to v9.1.0

```bash 
#!/bin/bash
 
# Create the download directory
cd ~
mkdir gcc_compile
cd gcc_compile
 
# Download and unzip GCC
wget https://ftp.gnu.org/gnu/gcc/gcc-9.1.0/gcc-9.1.0.tar.gz
tar xzf gcc-9.1.0.tar.gz
 
cd gcc-9.1.0
 
# Download dependencies
./contrib/download_prerequisites
 
# Create build directory
cd ..
mkdir gcc-9.1.0-build
cd gcc-9.1.0-build
 
# Configure Makefile
../gcc-9.1.0/configure --enable-languages=c,c++ --disable-multilib
 
# Compile and install
make -j4
sudo make install
 
# Remove GCC temp directories
cd ..
cd ..
rm -rf gcc_compile

closed 2019-11-12 by stale[bot]