espressif/arduino-esp32 #3062: Run Arduino example "BLE_client", reboot when connect BLE server

Belife2012, 2019-08-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
write the Sketch base on Arduino Example "BLE_client", BLE as client connect to a BLE server (is not esp32), use for receive NOTIFY. 
BLE client will keep scanning when server is not start. Then server start, Client reboot while connect to Server.  

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch【Use arduino example (BLE_client) 】
/**
   A BLE client example that is rich in capabilities.
   There is a lot new capabilities implemented.
   author unknown
   updated by chegewara
*/

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00008850-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0000885a-0000-1000-8000-00805f9b34fb");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
  Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
    }

    void onDisconnect(BLEClient* pclient) {
      connected = false;
      Serial.println("onDisconnect");
    }
};
BLEClient*  pClient;
BLERemoteService* pRemoteService;
bool connectToServer() {
  Serial.print("Forming a connection to ");
  Serial.println(myDevice->getAddress().toString().c_str());

  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  pClient->setClientCallbacks(new MyClientCallback());
[... 165 lines cut ...]
[D][BLERemoteCharacteristic.cpp:279] retrieveDescriptors(): Found a descriptor: Handle: 44, UUID: 00002902-0000-1000-8000-00805f9b34fb
 - Found our characteristic
[D][FreeRTOS.cpp:158] take(): Semaphore taking: name: ReadCharEvt (0x3ffe37ac), owner: <N/A> for readValue
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe37ac), owner: readValue
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_READ_CHAR_EVT
The characteristic value was: 
[D][FreeRTOS.cpp:158] take(): Semaphore taking: name: RegForNotifyEvt (0x3ffe3898), owner: <N/A> for registerForNotify
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: RegForNotifyEvt (0x3ffe3898), owner: registerForNotify
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d1f84: 20c0120c c0b14800 05380020
Core 1 register dump:
PC      : 0x400d1f88  PS      : 0x00060f30  A0      : 0x800d9d54  A1      : 0x3ffca340  
A2      : 0x3ffc506c  A3      : 0x3ffc4fc6  A4      : 0x00000020  A5      : 0x80000020  
A6      : 0x00000000  A7      : 0x3ffba2c4  A8      : 0x800d1f88  A9      : 0x3ffca2f0  
A10     : 0x00000001  A11     : 0xbab6c9c9  A12     : 0xbab6c9c9  A13     : 0x3ffc60ec  
A14     : 0x3ffc4fb4  A15     : 0x00fb0010  SAR     : 0x00000008  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d1f88:0x3ffca340 0x400d9d51:0x3ffca390 0x40090461:0x3ffca3b0

Rebooting...
```

Belife2012, 2019-08-03:
Use ESP Exception Decoder:

Decoding stack results
0x400d1f88: connectToServer() at E:\MCU\ESP32\Demo\BLE\BLE_client2/BLE_client2.ino line 92
0x400d9d51: loopTask(void*) at D:\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 19
0x40090461: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

wakwak-koba, 2019-10-02:
try my repo.
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE

closed 2019-12-16 by stale[bot]