espressif/arduino-esp32 #3065: BLE - Connecting to multiple peripherals one at a time not working

MohsinEngineer, 2019-08-03:
### Hardware:
Board:							DOIT ESP32 DevKit V1
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I want to connect to all the peripherals detected in a BLE scan ONE AT A TIME. Once connected, read/update characteristic values, disconnect and move on to the next one.
If there is only one peripheral in range, it works fine. But if 2 come in range with the same Service and characteristics etc. The client tries to establish connection to the next one even before the first one gets disconnected. This causes the code to get stuck. No error, just stuck in the serial monitor.

### Update:
For some reason, turning the Core debug level to debug makes the code run for more iterations than normally. Don't know how this is affecting the running the code but it is. It runs for 10-12 times then Serial monitor just stops just like in normal scenario.
I need the loop to run infinitely without risk of getting stuck like this. Any help will be highly appreciated.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "BLEDevice.h"
#include "BLEScan.h"

// The remote service we wish to connect to.
#define serviceUUID "51cc567a-e8c6-4af0-9315-e955c51ebe1d"
// The characteristics of the remote service we are interested in.
#define usernameUUID "e2322b44-d4b1-40bc-a95b-6d3a24537a62"
#define lastAccessUUID "7937b8b7-1692-4953-9c48-88838252f9ea"
#define deviceNameUUID "19A76C2D-DBBB-4A30-B927-991DCC75E5D8"

// Duration of scan in seconds
#define scanTime 5

static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEScan* pBLEScan;
std::string prevUsers[10];
uint8_t prevCount = 0;
static bool connected = false;

class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
      Serial.println(" - Connected to server\n");
      connected = true;
    }

    void onDisconnect(BLEClient* pclient) {
      Serial.println(" - Disconnected from server\n");
      connected = false;
    }
};

bool connectToServer(BLEAdvertisedDevice* currentDevice) {
  Serial.print("Forming a connection to ");
  Serial.println(currentDevice->getAddress().toString().c_str());

  BLEClient*  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  // Connect to the remote BLE Server.
  pClient->connect(currentDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
  Serial.println(" - Connected to server");

  pClient->setClientCallbacks(new MyClientCallback());

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(BLEUUID(serviceUUID));
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID);
    pClient->disconnect();
    return false;
[... 294 lines cut ...]
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe2bac), owner: readValue
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
Galaxy S8+
This device was last accessed [D][FreeRTOS.cpp:158] take(): Semaphore taking: name: ReadCharEvt (0x3ffe2624), owner: <N/A> for readValue
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe2624), owner: readValue
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
Time since boot: 7 seconds before
Setting new characteristic value to "Time since boot: 8"
[D][FreeRTOS.cpp:158] take(): Semaphore taking: name: WriteCharEvt (0x3ffe29c4), owner: <N/A> for writeValue
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: WriteCharEvt (0x3ffe29c4), owner: writeValue
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
We are now connected to the BLE Server.
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:611] removePeerDevice(): remove: 1, GATT role client
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:158] take(): Semaphore taking: name: ScanEnd (0x3ffdecf8), owner: <N/A> for start
[D][FreeRTOS.cpp:167] take(): Semaphore taken:  name: ScanEnd (0x3ffdecf8), owner: start

```

wakwak-koba, 2019-10-02:
try my repo.
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE

atanisoft, 2019-10-02:
@wakwak-koba maybe submit a PR with your changes since your repo may be out of date on other parts of the core...

wakwak-koba, 2019-10-02:
I merged to the latest 1.0.4 a few hours ago.

atanisoft, 2019-10-02:
@wakwak-koba great, but submitting a PR is best for the end user.

JuergenLeber, 2019-11-25:
Also just solved all my connection/disconnection problems with the version from @wakwak-koba. Please do a PR for that @wakwak-koba

closed 2020-02-07 by stale[bot]