espressif/arduino-esp32 #3066: compliing as esp-idf component, error compiling diskio.h after clean install

qt1, 2019-08-04:
With an old (working) project,

I have reinstalled esp-idf from repository
`   git describe --tags  =>   v4.0-dev-1381-g6fe853a2c`

and arduino-esp under components/arduino
`   git describe --tags  =>   1.0.3-rc1-42-gfa55a2c`

Now AzureIoT fails compilation, so I disabled the component in 'make menuconfig'

And now I get this king of error:
```
In file included from /home/baruch/esp/rmt_sender/components/arduino/libraries/SD/src/sd_diskio.cpp:16:0:
/home/baruch/esp/esp-idf/components/fatfs/src/diskio.h:13:9: error: 'BYTE' does not name a type
 typedef BYTE DSTATUS;
         ^
/home/baruch/esp/esp-idf/components/fatfs/src/diskio.h:29:1: error: 'DSTATUS' does not name a type
 DSTATUS disk_initialize (BYTE pdrv);

....

/home/baruch/esp/rmt_sender/components/arduino/libraries/WiFi/src/ETH.cpp:22:25: fatal error: eth_phy/phy.h: No such file or directory
compilation terminated.

```

It looks like the code tries to pull include files from esp-idf and there is some inconsistency between the version of the SDK

It may be something to do with the make system or some other ritual I didn't do right. However I am a bit concerned this indicates mismatch 
 
Please advise.

Thanks

lbernstone, 2019-08-04:
arduino-esp32 is currently built on esp-idf 3.2
https://github.com/espressif/arduino-esp32/issues/1142

qt1, 2019-08-05:
Following comment from @lbernstone (thanks!)

I Checked for revisions:
```
cd ~/esp/esp-idf
git branch -r
```
There are V3.0-V3.3 and the current - V4

Trying v3.2 like so:
`git checkout -b origin/release/v3.2 --recurse-submodules`

And after reverting to xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0. the compilation almost works.

This is the error I get:
```
/home/baruch/esp/rmt_sender/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, int, const char*, const char*, const char*, const char*, const char*)':
/home/baruch/esp/rmt_sender/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:146:67: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
                  (const unsigned char *)pskIdent, strlen(pskIdent));
```

I get the same error with v3.3 so I continued with V3.3 and added a crude patch:

line 9 of 
${workspaceRoot}/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp
```
#define MBEDTLS_KEY_EXCHANGE__SOME__PSK_ENABLED
```

And now it compiles and probably runs (needs some testing..)

This is probably something to do with configuration and I hope someone will fix or advise.

I have also looked a bit into V4' seems like there is a small change in the ip library that affects 4 files.

atanisoft, 2019-08-05:
You can also start with the sdkconfig from the arduino-esp32/tools/sdk directory. The WiFi client secure library requires a bit of the mbedtls bits to be enabled.

qt1, 2019-08-05:
arduino-esp32/tools/sdk is a duplication of parts of esp-idf

When working with esp-idf I prefer to relay on a single source - which is esp-idf
In the past I used to delete arduino-esp32/tools/sdk just to avoid confusion.

In any case it seems like compilation is good, for now.

Thanks.

closed 2019-08-05 by qt1

atanisoft, 2019-08-05:
@qt1 yes, it is a duplicate but there is a known good sdkconfig file there which you can use as a base for your IDF project. You don't need anything else from that tree