espressif/arduino-esp32 #3068: mcpwm_isr_register() compile error

mrguen, 2019-08-05:
I would like to test this mcpwm example script in the Arduino IDE 1.8.9, but it does not compile

https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c

1) This reference is not good:

#include "soc/mcpwm_periph.h"

 And this one should be added

 #include "soc/mcpwm_struct.h"

2) After un-comenting the capture configuration section there is an error on this line

mcpwm_isr_register(MCPWM_UNIT_0, isr_handler, NULL, ESP_INTR_FLAG_IRAM, NULL);  //Set ISR Handler

"invalid conversion from 'void (*)()' to 'void (*)(void*)' [-fpermissive]"

lbernstone, 2019-08-05:
arduino-esp32 is currently using esp-idf v3.2.  If you use the example from that version, I think it will work properly.

mrguen, 2019-08-06:
Thanks for this answer. 
I can't see an esp-idf-3.2 examples directory.
Is there a documentation for Arduino implementation of isp-idf-3.2?

lbernstone, 2019-08-06:
https://github.com/espressif/esp-idf/blob/release/v3.2/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c
[esp-idf documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/mcpwm.html)

mrguen, 2019-08-06:
Thank you very much!

closed 2019-10-19 by stale[bot]