espressif/arduino-esp32 #3072: ESP32 Board not found on Arduino IDE 1.8.9

syswombat, 2019-08-06:
### Hardware:
Board:							Raspberry Pi 4 4GB RAUM
Core Installation version:			Buster , Debian 10
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        Debian 10

### Description:
Hello

i followed the step there

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

and the second one which start with
cd ~/Arduino/hardware

at the end 
***
:~/Arduino/hardware/espressif/esp32/tools# python3 get.py
System: Linux, Info: Linux-4.19.58-v7l+-armv7l-with-debian-10.0
Platform: arm-linux-gnueabihf
Tool esptool-2.6.1-linux.tar.gz already downloaded
Extracting esptool-2.6.1-linux.tar.gz
Tool mkspiffs-0.2.3-arduino-esp32-linux-armhf.tar.gz already downloaded
Extracting mkspiffs-0.2.3-arduino-esp32-linux-armhf.tar.gz
Renaming mkspiffs-0.2.3-arduino-esp32-linux-armhf to mkspiffs
Done
***
no error nothing the download did happen, all fine - BUT ... 
do i start the Arduino IDE the board esp32 is not showing up so what should be wrong?

have a nice day
Vinc

lbernstone, 2019-08-06:
https://github.com/espressif/arduino-esp32/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+raspberry

syswombat, 2019-08-07:
@lbernstone your added linke seems not to show a solution, or did I miss it?

atanisoft, 2019-08-07:
@syswombat try https://github.com/espressif/arduino-esp32/issues/789.. it should be possible for running the generic Linux install by cloning the git repo to the ~/Arduino/hardware/esp32 directory and copy the gcc bits to the tools directory.

lbernstone, 2019-08-07:
Your question has been answered many times.  Look through the answers there.  Raspberry Pi is not pre-compiled ATM.

syswombat, 2019-08-16:
Hardware Raspberry Pi 5 with Debian Buster 10
-----------------------------------------------------
Hello

to this installation i took the second one
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

doing a search for the directory i have
/root/Arduino/hardware
/opt/arduino-1.8.9/hardware
/home/pi/.arduino15/packages/arduino/hardware

i did it on 
/root/Arduino/hardware

down at
~/Arduino/hardware/espressif/esp32/tools
# python3 get.py

System: Linux, Info: Linux-4.19.58-v7l+-armv7l-with-debian-10.0
Platform: arm-linux-gnueabihf
Tool esptool-2.6.1-linux.tar.gz already downloaded
Extracting esptool-2.6.1-linux.tar.gz
Tool mkspiffs-0.2.3-arduino-esp32-linux-armhf.tar.gz already downloaded
Extracting mkspiffs-0.2.3-arduino-esp32-linux-armhf.tar.gz
Renaming mkspiffs-0.2.3-arduino-esp32-linux-armhf to mkspiffs
Done

but still the esp32 boards are not found by Android IDE

lbernstone, 2019-08-16:
Did you read https://github.com/espressif/arduino-esp32/issues/789?

syswombat, 2019-08-16:
yes i did, but 

the Toolchain should be working as describe her 
https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/linux-setup-scratch.html
with some modification about bash 5 and ccg8 to ccg6

your link is about Raspberry 3 with Debian 9
i have a Raspberry Pi 4 with Debian 10

syswombat, 2019-09-16:
still open the problem

closed 2019-11-29 by stale[bot]