espressif/arduino-esp32 #3076: ESP32 OTAWebUpdater using softAP doesn't work

amiilto, 2019-08-08:
Board: ESP-WROOM32
Core Installation/update date: Latest
IDE: Arduino IDE
Flash Frequency: 80Mhz/40Mhz
Upload Speed: 115200
Computer OS: Windows 10

So, I'm creating a OTA update server using the OTAWebUpdater example. When I connect it to my WiFi router, it works perfectly. The problem is, if I create a softAP instead of connecting it to my WiFi, it doesn't work anymore. The browser window returns the following:

"Not found: /serverIndex".

In the code, the only changes I made were WiFi.softAP(ssid, pass) instead of WiFi.begin(ssid, pass), and I removed the "wait for connection" loop.

What should I do?

lbernstone, 2019-08-08:
Did you remark out the wait for connection part?

amiilto, 2019-08-08:
> Did you remark out the wait for connection part?

Yes, as soon as I noticed that it didn't work I tried without removing it and it still didn't work.

lbernstone, 2019-08-08:
The example works for me.  Please turn on verbose logging under the tools menu, flash the program again, and post the log from serial monitor. Check that you can ping the device at 192.168.4.1 from your PC after you connect to the SoftAP.

closed 2019-10-21 by stale[bot]

musalmz, 2022-02-25:
I am having the same problem.  Can you help me pls?

italocjs, 2022-03-10:
I think its related to the inclusion of a external code 
`"<script src='https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js'></script>"`
since there is no internet available, this fails and it responds that the serverindex handler was not found.   I'm not familiar with websites programming and have no idea on how to fix this.

italocjs, 2022-03-10:
I managed to fix the issue thanks to a old post from "Machiel" (link: https://esp32.com/viewtopic.php?t=11744)

Here is the full, working code.

`#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>

const char *host = "esp32";
//const char *ssid = "SimovaSkyNet";
//const char *password = "pr0dut1v1d4d3";

const char *ssid = "SimovaMPA_5036";
const char *password = "12345678";

WebServer server(80);

/*	jquery.min.js version 3.2.1

	size of original js file:  85 KB
	size in this source file: 148 KB
	size on ESP:               30 KB
 
 */
#define jquery_min_js_v3_2_1_gz_len 30178
PROGMEM const char jquery_min_js_v3_2_1_gz[]  = {
0x1f,0x8b,0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x0a,0xcc,0xbd,0x69,0x77,0x1b,0x47,0x92,0x36,0xfa,0xfd,0xfd,0x15,0x44,0xb5,0x86,0x5d,0x25,0x24,0x41,0x50,0xb6,0xfb,0x9d,0x2e,0xa8,0x84,0x63,0x5b,0x56,0x5b,0x3d,0xde,0xda,0x92,0x97,0x1e,0x10,0xf6,0xa9,0x0d,0x40,0x81,0xd8,0x08,0x80,0xa2,0x64,0x02,0xf3,0xdb,0x6f,0x3c,0x11,0x99,0x59,0x59,0x0b,0x68,0xf7,0xcc,0x9d,0x73,0xaf,0x17,0xa2,0x96,0xac,0x5c,0x23,0x23,0x63,0x8f,0xcb,0xa7,0x9d,0xb3,0xf9,0x3f,0xee,0xf2,0xed,0x87,0xb3,0x77,0x1f,0xf5,0x9e,0xf5,0xae,0xce,0x0e,0x67,0x7e,0x1a,0x9c,0xfd,0xfd,0xcd,0xd9,0xab,0xf5,0xdd,0x2a,0x8b,0xf7,0xc5,0x7a,0x75,0x16,0xaf,0xb2,0xb3,0xf5,0x7e,0x96,0x6f,0xcf,0xd2,0xf5,0x6a,0xbf,0x2d,0x92,0xbb,0xfd,0x7a,0xbb,0xa3,0xa2,0xf3,0x5b,0x7c,0xda,0x5b,0x6f,0xa7,0x97,0x8b,0x22,0xcd,0x57,0xbb,0xfc,0xec,0xe9,0xe5,0xff,0xe9,0x4c,0xee,0x56,0x29,0x3e,0xf4,0x63,0x95,0x04,0x0f,0xde,0x1d,0x3d,0xde,0xd1,0x67,0xe9,0xde,0x1b,0x78,0xeb,0x64,0x9e,0xd3,0x45,0x14,0xed,0x3f,0x6c,0xf2,0xf5,0xe4,0x6c,0xb9,0xce,0xee,0x16,0xf9,0xf9,0xf9,0x89,0x17,0xbd,0xfc,0xfd,0x66,0xbd,0xdd,0xef,0x86,0xd5,0xdb,0x28,0xee,0x65,0xeb,0xf4,0x6e,0x99,0xaf,0xf6,0xc3,0x84,0x9a,0xe9,0xf4,0x83,0xb0,0x6c,0x35,0x78,0x28,0x26,0x7e,0xa7,0x2c,0x12,0xec,0x67,0xdb,0xf5,0xfd,0xd9,0x2a,0xbf,0x3f,0xfb,0x62,0xbb,0x5d,0x6f,0x7d,0x4f,0x8f,0x79,0x9b,0xdf,0xde,0x15,0xdb,0x7c,0x77,0x16,0x9f,0xdd,0x17,0xab,0x8c,0xca,0xdc,0x17,0xfb,0x19,0xdd,0x99,0x2f,0xbd,0x60,0xb0,0xcd,0xf7,0x77,0xdb,0xd5,0x19,0xb5,0x12,0x1c,0x43,0xfe,0xeb,0x7b,0x34,0x33,0xf9,0xa4,0x58,0xe5,0x99,0xd7,0x31,0xdd,0x95,0xef,0x87,0xf2,0x13,0xee,0x67,0xc5,0x4e,0x3d,0x32,0x0d,0xef,0x62,0x9a,0xcb,0x68,0x34,0x56,0x99,0x33,0x12,0x95,0x47,0xdf,0xf2,0x24,0xf4,0xa6,0xf9,0xfe,0xbb,0xed,0x7a,0xbf,0x46,0xdd,0xdf,0x4e,0xd4,0x24,0x4a,0x7b,0x3b,0x4c,0xb0,0x9a,0xd2,0x15,0xad,0x41,0x1a,0xef,0xd5,0x8c,0x2e,0x37,0x77,0xbb,0x99,0x2a,0xe8,0x82,0x5a,0xcd,0xdf,0x53,0xc9,0x79,0xf4,0x70,0x54,0x37,0xd1,0xbc,0xb7,0x5f,0xbf,0xa1,0xa6,0x56,0x53,0xb5,0xa0,0x9b,0x59,0xbc,0xfb,0xf6,0x7e,0x45,0x35,0x6e,0xf2,0xed,0xfe,0x83,0x5a,0x46,0x8b,0xf2,0xfd,0x2a,0x5a,0xf6,0xd2,0x78,0xb1,0xf0,0xa5,0xe9,0x40,0xad,0xa9,0x8a,0x81,0xe9,0xfa,0xd9,0x46,0x3a,0x9f,0x44,0xc9,0xe1,0x90,0xe9,0x6e,0x27,0xbd,0x74,0x9b,0xc7,0xfb,0xfc,0x8b,0x45,0x8e,0x6e,0xfb,0xde,0x2e,0xdd,0x16,0x1b,0xcc,0x55,0xda,0xdb,0xe7,0xef,0xf7,0x11,0x7d,0xd2,0x9b,0xe5,0x71,0xd6,0x8b,0x37,0x9b,0x7c,0x95,0x7d,0x3e,0x2b,0x16,0x19,0x41,0x55,0x6f,0x13,0x6f,0xa9,0xfc,0x37,0xeb,0x2c,0xef,0x6d,0xf3,0xe5,0xfa,0x5d,0x6e,0xde,0x1c,0x51,0xf1,0x6d,0xe4,0x31,0x08,0x7a,0x6a,0x1b,0x55,0xa7,0x4e,0x2f,0x01,0xd6,0x6f,0xdb,0x9b,0xac,0x68,0xb4,0xc5,0x9e,0xdf,0x1c,0xd5,0x2e,0xba,0xfc,0x65,0x74,0xbd,0xbb,0xbe,0x7b,0xf5,0xc5,0xab,0x57,0xd7,0xef,0x3f,0xed,0x8f,0xbb,0x87,0xda,0xfd,0x93,0xcb,0xa9,0xda,0x53,0xb1,0x8b,0xe5,0xee,0xe2,0x52,0xdd,0x45,0x97,0x17,0xfe,0x28,0xbe,0xf8,0x6d,0x1c,0xd0,0xf3,0x77,0xed,0x2d,0x25,0x34,0x3f,0x3f,0x50,0xdf,0xb7,0x9f,0xc7,0xbb,0xdc,0x0f,0x8e,0x03,0x34,0x1b,0x6d,0x7b,0x1b,0xb3,0x28,0xd1,0x83,0x80,0x7e,0x78,0xab,0x68,0x39,0x68,0x55,0xef,0x52,0xda,0x12,0xe1,0x56,0x2d,0xf2,0xd5,0x74,0x3f,0x0b,0xfb,0x6a,0xbf,0xfe,0x74,0xbb,0x8d,0x3f,0x94,0x50,0x69,0x2b,0x9f,0xc8,0x7c,0x03,0x44,0x68,0x00,0xb4,0xd4,0x15,0xc8,0x35,0x63,0xbd,0x5b,0x2c,0xa2,0x28,0x1e,0xba,0x85,0xc3,0xf8,0x79,0x7f,0x88,0xab,0x51,0xdc,0xc5,0x4f,0x4f,0x1a,0x1b,0x87,0xf2,0x6c,0x7c,0x54,0x80,0x87,0x37,0xfb,0x38,0xbd,0xa9,0x54,0x89,0xc9,0x4d,0xa8,0xf7,0xcb,0x7c,0x3b,0xcd,0xb9,0xaa,0x9e,0xd3,0x69,0x3f,0x50,0x71,0x09,0xe5,0x34,0xc4,0xfc,0x9d,0x80,0x42,0xc4,0x40,0x9c,0x1c,0x55,0x1e,0xa7,0xb3,0xb6,0x3e,0x6e,0x7b,0x78,0xc3,0x15,0x52,0x15,0x47,0xb5,0x8c,0x37,0x6d,0xc5,0xb8,0x41,0xdb,0x33,0x9f,0xfa,0x11,0x6f,0xfc,0xea,0x06,0x49,0x54,0x6a,0x8b,0xc7,0x32,0x62,0x7a,0x84,0x05,0x0e,0xb0,0xc6,0x00,0xfd,0x96,0x89,0xac,0x55,0x3c,0x01,0xb8,0x2d,0x3e,0xe8,0xfe,0x6c,0xa7,0xbc,0xa7,0x76,0xa8,0x60,0x52,0x6c,0x77,0xfb,0x53,0x15,0xe4,0xb7,0x7e,0x9f,0xca,0x2c,0xe2,0x47,0x8b,0x5c,0x5c,0x51,0x99,0xfc,0xb6,0x65,0x5e,0x9d,0x95,0x50,0x69,0xd4,0x8d,0xbb,0x3e,0x96,0x29,0x09,0xfb,0x76,0x52,0x6b,0xfd,0x4c,0x5f,0x44,0xfd,0xf3,0xf3,0xf4,0x79,0x32,0x1c,0xf1,0xc2,0xa5,0xe3,0x71,0x38,0x1a,0xa3,0xfa,0x55,0x76,0x72,0x94,0x76,0x55,0x0e,0x87,0xe6,0x02,0xca,0xc2,0x87,0x33,0xb5,0x23,0xdc,0x18,0x12,0xae,0xa0,0x1f,0xb5,0xdb,0xf0,0xb4,0xd1,0x1d,0x5f,0x1c,0x15,0x2d,0xd7,0xfb,0x3d,0xb5,0x11,0xf1,0x1e,0xd2,0xd7,0x4e,0x7b,0x18,0x0e,0x6d,0x02,0x9a,0xf7,0x4c,0xe5,0x6a,0x42,0xb8,0xc6,0x4e,0xe2,0xa8,0x3f,0x3e,0x1c,0x08,0xb3,0xcc,0xa2,0x2b,0x42,0x37,0xf6,0xb1,0x19,0xf6,0x3c,0xea,0x5c,0x0d,0x26,0xc0,0xab,0xc9,0x7a,0xbd,0xc8,0xe3,0x55,0x89,0xc5,0xa7,0xe7,0xe7,0xfe,0x3c,0x9a,0x56,0x2a,0x9b,0xe9,0xca,0xba,0xdd,0x40,0x35,0xd0,0xfe,0xf4,0x70,0xd8,0xf6,0x8a,0xdd,0x2b,0xd3,0xaf,0x69,0x70,0x38,0xf8,0x53,0xc2,0x49,0x01,0xb5,0x1e,0x45,0x05,0xd5,0x37,0x15,0xc8,0x9c,0x5d,0x5c,0x04,0x83,0xd9,0xf3,0x62,0x80,0x8a,0x08,0xe1,0x63,0xcb,0x74,0x22,0x3f,0xae,0xb4,0 [... cut ...]
            }
            else if (upload.status == UPLOAD_FILE_END)
            {
                if (Update.end(true))
                { // true to set the size to the current progress
                    Serial.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
                }
                else
                {
                    Update.printError(Serial);
                }
            }
        });
    server.begin();
}

void loop(void)
{
    server.handleClient();
    delay(1);
}`

softvaree, 2022-03-23:
Thank you very much, the code works fine.

R0binBl00d, 2024-04-01:
nice one, really love it. Was struggling with the "serverindex cannot be found" for quite some time ...

this even helped me with my SoftAP-approach.
But I didn't use the custom style or Strings instead of const char* ...

the only update required was the "PROGMEM",  "script src=", "void onJavaScript" and the "server.on("/jquery.min.js", [...]);"

works perfectly without an external Wifi-Router :-)
Thanks again

`#define jquery_min_js_v3_2_1_gz_len 30178`
`PROGMEM const char jquery_min_js_v3_2_1_gz[]  = {`
[...]
`};`

`const char* serverIndex = `
`"<script src='/jquery.min.js'></script>"`
[...]
`"</script>";`

`void onJavaScript(void) {`
`    Serial.println("onJavaScript(void)");`
`		server.setContentLength(jquery_min_js_v3_2_1_gz_len);`
`		server.sendHeader(F("Content-Encoding"), F("gzip"));`
`    server.send_P(200, "text/javascript", jquery_min_js_v3_2_1_gz, jquery_min_js_v3_2_1_gz_len);`
`}`

`/* setup function */`
`void setup(void) {`
`  Serial.begin(115200);`

`  // Connect to WiFi network`
`  WiFi.softAP(ssid, password);`
`  Serial.println("");`

`  Serial.print("Connected to ");`
`  Serial.println(ssid);`
`  Serial.print("IP address: ");`
`  Serial.println(WiFi.softAPIP());`

`  /*use mdns for host name resolution*/`
`  if (!MDNS.begin(host)) {`
`    Serial.println("Error setting up MDNS responder!");`
`    while (1) {`
`      delay(1000);`
`    }`
`  }`
`  Serial.println("mDNS responder started");`
`     /*return javascript jquery */`
`  server.on("/jquery.min.js", HTTP_GET, onJavaScript);`
[...]
`}`