espressif/arduino-esp32 #3079: [E][esp32-hal-i2c.c:1146] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init

HannHank, 2019-08-10:
Hey,
I have a problem with my esp32 and the i2C. I am building a car and I need to read out the compass and the accelerometer Data. On the Arduino Uno everything worked fine, but on the esp32 this error appears:

`[E][esp32-hal-i2c.c:1146] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init
`

Do I have to add a 4.7kΩ pull-up resistors to both lines ?

thanks in advanced.

stickbreaker, 2019-08-10:
@HannHank Yes, you need pullups.  the ESP32's built in pullups are too weak for I2C.  Since the Esp32 is 3.3v  I would recommend 3.3k ohm.  This provides 1mA of pullup.  It will allow 400kHz maximum bus speed.

Chuck.

HannHank, 2019-08-10:
Thank you for your quick answer. But I am confused, because I saw so many tutorials where people just connected the SDA and the SCL to the esp32 and everything worked fine. It is just because mine got to weak pull-ups ?

stickbreaker, 2019-08-10:
@HannHank The esp32's pullups are Weak.  They are not spec'd in the datasheet. They can range from 10k ohm to over 100k ohm.  They vary by port and production.  The [I2C specification UM10204 Philips/NXP](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) lists the maximum signal rise time for both 100kHz and 400kHz bus. for 100kHz it is 1us, for 400kHz it is 300ns.  The combination of pullup value and bus capacitance.  With the weak, unspecified pullups of the esp32 it is not recommend to run without pullups.  Best practice it to use pullups.  There is no enforcement, it may work.  I don't build/design system that "may work".  It is your choice.  If you want to run bareback, you should at least measure your bus capacitance and signal waveform.  To me, the partial penny you save by not including pullup resistors would not not be worth my time.

Here is a wiki I created discussing pullup value considerations [i2c Pullups](https://github.com/stickbreaker/arduino-esp32/wiki/Pullup-Values).

Chuck.

HannHank, 2019-08-11:
@stickbreaker thank you for your help, 
I added two 4.7 ohm pullups to my circuit and everything worked fine. But I later figured out that the issue: 

`[E][esp32-hal-i2c.c:1146] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init`

just appears if you upload the Code to the esp32 and open the Serial Monitor. The esp32 in my case needed a complete reconnect. I don't now why but this fixed it.

closed 2019-08-11 by HannHank

cyberman54, 2019-10-15:
@stickbreaker sorry for hitchhiking this thread. I have a similar poblem after ESP32 wakes up from deep sleep i find the i2c bus unusable:

```
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0

```
Thus i can't re-init the i2c slaves. Is there any special treatment for the SDA/SCL GPIOs required after ESP32 wakes up from deep sleep? I am stuck here and did not find help at Google for this issue, so any hint would be appreciated.

Davincible, 2020-06-27:
@HannHank your suggestion also just fixed my issue. Which makes me wonder, is there any fix to make it work without cutting power?

mh4nx7net, 2021-01-08:
sometimes, it's hapen when your slave/master devace current are less from what the actuall current device voltage need

hope's help

federic1, 2021-06-01:
> sometimes, it's hapen when your slave/master devace current are less from what the actuall current device voltage need
> 
> hope's help

This solution works!! Thank you