espressif/arduino-esp32 #3084: ESP_32 BT CLASSIC : SerialBluetooth lib. as MASTER : SCAN AND CONNECT TO CLIENT

rommo911, 2019-08-12:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module WROOM-32
Core Installation version:			1.0.0
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:				      115200
Computer OS:                                        Windows 10
### Description:
Describe your problem here

 hello every one , this is my first post over here
i am new to the ESP-32 : i want to use it as serial bt connection : BT CLASSSIC + Master role : scan an connect to client : search by name or by MAC and connect.
as far as i know the serialbluetooth library does the trick but only in slave mode as server 
i ran some research and use the library code + code from idf ide example (SPP_initiator_example) 
the search by bt name failed i will post the failure in another post , but with a trick to search only by mac it does work:
what i get is search  + compare mac add + connect to client but NO CONNECTION IS OPEN 
tried to connect to my phone, my pc and an OBDII adapter 
in connect succefully and with pin code authinticate and code authinticating success all is done
 but i am no getting a connection established : waiting for  
ESP_SPP_OPEN_EVT://Client connection open
        log_i("ESP_SPP_CLINT_CONNECT_EVT");
but not getting it 
i am sure i am missing something over here maybe you help me 
thank all .

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

// Copyright 2018 Evandro Luis Copercini
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include "sdkconfig.h"
#include <cstdint>
#include <cstdio>
#include <cstdlib>
#include <cstring>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "time.h"
#include "sys/time.h"

#if defined(CONFIG_BT_ENABLED) && defined(CONFIG_BLUEDROID_ENABLED)

#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
[... 516 lines cut ...]
    custom_spp_callback = callback;
    return ESP_OK;
}

#endif
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
14:45:18.613 -> [I][BluetoothSerial_edited.cpp:189] esp_spp_cb(): ESP_SPP_INIT_EVT
14:45:18.613 -> [I][BluetoothSerial_edited.cpp:191] esp_spp_cb(): ESP_SPP_INIT_EVTset mode
14:45:18.613 -> [I][BluetoothSerial_edited.cpp:276] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
14:45:18.852 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:18.921 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:18.955 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:19.195 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:19.229 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:22.217 -> [I][BluetoothSerial_edited.cpp:266] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
14:45:22.217 -> [I][BluetoothSerial_edited.cpp:276] esp_bt_gap_cb(): SP_BT_GAP_DISC_STATE_CHANGED_EVT
14:45:24.460 -> [I][BluetoothSerial_edited.cpp:240] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
14:45:24.460 -> [I][BluetoothSerial_edited.cpp:242] esp_spp_cb(): ESP_connect
14:45:24.460 -> [I][BluetoothSerial_edited.cpp:254] esp_spp_cb(): ESP_SPP_CL_INIT_EVT
14:45:24.494 -> [I][BluetoothSerial_edited.cpp:246] esp_spp_cb(): ESP_SPP_CLINT_CONNECT_EVT

hhk7734, 2019-09-18:
I'm not sure if this is related to 'sdkconfig.h'.
But, I think #2320 helps.

SelaliAdobor, 2019-09-21:
https://github.com/SelaliAdobor/BTClassicSPP

Created a PR for the BTClassicSPP library based off this repo that adds support for master with a fixed pin (adding a variable pin would be pretty straight forward)

hhk7734, 2019-09-21:
> https://github.com/SelaliAdobor/BTClassicSPP
> 
> Created a PR for the BTClassicSPP library based off this repo that adds support for master with a fixed pin (adding a variable pin would be pretty straight forward)

is it tested?

SelaliAdobor, 2019-09-23:
Not sure what you mean

closed 2019-12-07 by stale[bot]