espressif/arduino-esp32 #3087: How do I upload ESP8266 Sketch Data?

Buzzardbait, 2019-08-13:
When using the Arduino IDE 1.8.9  when I try to upload my index.html file to SPIFFS I don't get the "ESP8266 Sketch Data Upload" in the Tools menu. Instead, I get "WiFi101/WiFiNINA Firmware Updater". Do I need to link to a certificate to get the "ESP8266 Sketch Data Upload"? How do I do that? Of course, my sketch compiles ok but I get "404 Error File Not Found" in my browser because it was never in the SPIFF for it to be sent to the browser.

In my sketch I'm using:
String getContentType(String filename); // convert the file extension to the MIME type
bool handleFileRead(String path); // send the right file to the client (if it exists)
SPIFFS.begin(); // Start the SPI Flash Files System
  server.onNotFound([]() { // If the client requests any URI
  if (!handleFileRead(server.uri())) // send it if it exists
  server.send(404, "text/plain", "404: Not Found"); // otherwise, respond with a 404(Not Found)error
});

lbernstone, 2019-08-13:
Please post esp8266 questions on the esp8266 forum.

Buzzardbait, 2019-08-13:
Ok

closed 2019-08-13 by Buzzardbait