espressif/arduino-esp32 #3088: Web authentication

amiilto, 2019-08-14:
Hello!

I'm trying to create some sort of login/authentication in my web server. I used the WebServer.h library in other project and I succesfully created an authentication box using the following:

`server.on("/", {
if(!server.authenticate(www_username, www_password))
return server.requestAuthentication();
server.send(200, "text/html", serverIndex);
});`

The problem is, I'm using the ESPAsyncWebServer.h library in this new project and I haven't found something similar. Is there any way to do it? Thank you!

atanisoft, 2019-08-14:
How about file the issue over on [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer).... Actually looks like you did! https://github.com/me-no-dev/ESPAsyncWebServer/issues/573

You can close this one and track it over there instead as it is unrelated to arduino-esp32!

amiilto, 2019-08-14:
I solved the problem with the following:

`server_async.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    if(!request->authenticate(www_username, www_password)) {
      return request->requestAuthentication();
    }
    request->send(200, "text/html", "index.html");                                          
  });`

closed 2019-08-14 by amiilto