espressif/arduino-esp32 #3089: `VFSImpl::open(char const*, char const*)': ?

ThatBigPrint, 2019-08-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                       no
Upload Speed:					115200
Computer OS:                                           Mac OSX

### Description:
No errors showing up in code it just wont compile new to the whole programing thing....

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <Wire.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

#define ADC_BITS    12
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1106_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
//U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
//U8G2__MIDAS_128X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB13, /* data=*/ PB15, /* cs=*/ PB12, /* dc=*/ PB14, /* reset=*/ PA8);

int count = 0;
const int cycleDelay = 20;      // Change to adjust cycle time

// offset var (multiply by 100)
int offset = 0;
String stat;
const int numReadings = 10;
int mode = 0;
// 1 = Bar
// 2 = PSI
// 3 = kPa

int Bar = 0;
int PSI = 0;
int kPa = 0;

int boostPressure;
int boostMax = 0;
int boostMin = 0;

int readings[numReadings];
int readIndex = 0;
int total = 0;
int average = 0;

unsigned long startMillis;
unsigned long currentMillis;
const unsigned long period = 5;

const int sensorHistoryLength = 128;
int sensorHistory[sensorHistoryLength];
int sensorHistoryPos = sensorHistoryLength - 1;

bool injector = false;          // are we using injectors
float ONCycle = 0;              // oncycle variable 
float OFFCycle = 0;             // offcycle variable got microsecond
float T = 0;                    // tota l time to one cycle ONCycle + OFFcycle
int F = 0;                      // Frequency = 1/T
int RPM = 0;                    // RPM (Frequency * 60)
float DutyCycle = 0;            // D = (TON/(TON+TOFF))*100 %
[... 393 lines cut ...]
}
```

### Debug Messages:

Arduino: 1.8.5 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

Archiving built core (caching) in: /var/folders/1g/l5z3234906n89cm_j4b7n5g00000gn/T/arduino_cache_591637/core/core_esp32_esp32_esp32doit-devkit-v1_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_2f8f9d204ff7b886536d860ab572131a.a
libraries/FS/vfs_api.cpp.o: In function `VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*)':
/Users/chris/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:127:(.text._ZN11VFSFileImplC2EP7VFSImplPKcS3_+0x62): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: stat
libraries/FS/vfs_api.cpp.o: In function `VFSFileImpl::_getStat() const':
/Users/chris/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:127:(.text._ZNK11VFSFileImpl8_getStatEv+0x3e): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: stat
libraries/FS/vfs_api.cpp.o: In function `VFSImpl::open(char const*, char const*)':
/Users/chris/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:127:(.text._ZN7VFSImpl4openEPKcS1_+0x52): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: stat
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

atanisoft, 2019-08-14:
looks similar to https://github.com/espressif/arduino-esp32/issues/2975 and https://github.com/espressif/arduino-esp32/issues/1422

ThatBigPrint, 2019-08-14:
> looks similar to #2975 and #1422

i have already looked at #2975 and i dont have any time mentioned in the code ?

atanisoft, 2019-08-14:
It would appear that TFT_eSPI will include ["SPIFFS.h"](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L123) note that it uses " for the include which means that it will scan your library tree for a SPIFFS.h and not guaranteed to be the one that is included in Arduino15/packages/esp32/libraries/SPIFFS.

atanisoft, 2019-08-14:
Check if you have a rogue version of SPIFFS.h under your arduino libraries tree (not the esp32/libraries/SPIFFS version)

ThatBigPrint, 2019-08-14:
there seams to be files with simular name in the tft file ?

ThatBigPrint, 2019-08-14:
shal i delete the non esp32 one ?

atanisoft, 2019-08-14:
You can try that and it may fix it.

ThatBigPrint, 2019-08-14:
deleted but still same error maybe i need to install SPIFFS.h ?

atanisoft, 2019-08-14:
SPIFFS is already installed with the arduino-esp32 stack. There is another library installed which is providing "time.h" and that is causing the problem.

ThatBigPrint, 2019-08-14:
ok i will try and find that will i need to delete that ?

lbernstone, 2019-08-14:
Enable the verbose debugging during compilation, and look through it carefully to see if it is using libraries from a location other than the esp32 root (other than your tft_espi.h).

ThatBigPrint, 2019-08-14:
Using library SPI at version 1.0 in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/SPI 
Using library TFT_eSPI at version 1.4.16 in folder: /Users/chris/Documents/Arduino/libraries/TFT_eSPI 
Using library FS at version 1.0 in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/SPIFFS 
Using library Wire at version 1.0.1 in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/Wire

ThatBigPrint, 2019-08-14:
looks to be using the correct one to me ?

lbernstone, 2019-08-14:
Something is damaged in your installation. I'd uninstall arduino-esp32 in board manager and then reinstall. If you just want to get around the problem, remark `#define SMOOTH_FONT` in User_setup.h

ThatBigPrint, 2019-08-14:
that has worked for now i have tried to re install but its still broken :/

closed 2019-10-27 by stale[bot]

Den2016, 2022-07-01:
Reason found. Variable `stat`. Just rename in your sketch in Stat or change to other name - and compiles successfull. 
In my sketch i using `byte stat` and get same error, renaming solves this problem