espressif/arduino-esp32 #3090: esp32 arduino esp_camera.h

msj121, 2019-08-14:
I have tried the latest arduino and all boad manager versions of esp23 (1.0.0,1.0.1,1.0.2).

The example for webserver doesn't show up in the examples beside all the other standard examples (like AnalogOut folder). It is not in the physical location when I navigate outside of Arduino IDE either.

I copied the code (https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino) manually to a file, tried to compile, but now:

> esp_camera.h: No such file or directory

I cannot find this header file in the current repository, but I did find it in the esp-camera repository, https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h

Am I missing something? I have uninstalled Arduino and ESP32 and reinstalled without any luck. I am running OS 10.14.4, I have tried the latest Arduino and also 1.8.6.

Very confused, thanks for any help.

atanisoft, 2019-08-15:
The camera example should be available in the Arduino IDE under Examples -> ESP32 -> Camera -> CameraWebServer.

You can not use the "master" version as it doesn't match up to any of the releases you have referenced. You can install the Development release from [here](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) and select the 1.0.3-rc1 version. Be sure to cleanup your existing versions prior to installing via boards manager as Arduino IDE may not clean them up fully.

msj121, 2019-08-15:
Oh WOW! Thank you! I was using the standard json. I spent all day on this before asking for help, so thank you so much.

closed 2019-08-15 by msj121

eylr, 2024-03-15:
once It happened to me for the stupidest reason imaginable.
The definition at Arduino IDE 2.3.2 should be: 
Tools -> Board   "AI  thinker esp32-cam " (it changed from some reason)