espressif/arduino-esp32 #3091: ESP32 reload when Fan is power on

mmashraf, 2019-08-15:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-08-15:
Please follow the [issue template instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md)

mmashraf, 2019-08-16:
Ibernstone: describe as advise. futher looking forward for response from people. as I come to know that ESP32 is very sensitive to AC voltage

lbernstone, 2019-08-16:
These errors look like code problems, not power (brownout) issues.  Try a simple blink sketch and see if you have the same errors.

mmashraf, 2019-08-17:
There is no error if ESP32 is running code except when // Assign output variables to GPIO pins
const int output26 = 19; as given in code. I have changed Pin to other Pins but issue remains same , this problem is related to 220v Fan control via relay module.  if I run LED bulb etc on same pin, there is no issue, it works fine with with any error.

lbernstone, 2019-08-17:
This forum is for problems with the code, not electrical issues.
You need to add capacitors if you have voltage issues, or a diode if you have backflow EMF.

mmashraf, 2019-08-17:
I understand but above errors are related to ESP32 code issues. further please advise where to add capacitor(value and where to put it) or diode suitable for it. Further I have already added 1N5817 schottky diode in  between relay module input Pin and coming power pin of esp32 to relay module

lbernstone, 2019-08-17:
If the exact same code works without the relay connected, then it is not a code error.  You are not explaining your problem well.
[flyback diode](https://www.google.com/search?q=flyback+diode+relay+arduino)
I cannot size your capacitor, since I don't have any idea what your system looks like (and I don't care to).  Please use google instead of wasting our time here.

closed 2019-10-30 by stale[bot]

parth2326, 2020-10-19:
Has anyone find a solution for it. I got the same issue when I control AC fan some time ESP32 module restart or hang. How can I reduce electromagnetic interference of fan operation to work properly?