espressif/arduino-esp32 #3093: Guru Meditation Error: Core  1 panic'ed (StoreProhibited)

kbickham, 2019-08-15:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.  

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							nodemcu 32s
Core Installation version:			1.02 arduino boards manager
IDE name:						?Arduino IDE? 
Flash Frequency:					?40Mhz?	
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? 

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <Arduino.h>
#include <genieArduino.h>
#include <ThingSpeak.h>
#include <HX711_ADC.h>
#include <Adafruit_Sensor.h>
#include <HardwareSerial.h>   
#include <AS726X.h>
#include <WiFi.h> //change to local if issues
#include <HardwareSerial.h>   
#include <ArduinoJson.h> 

//MLX ADDRESSING VALUES
#define MLX1AD 0x5A
// RAM
#define RAWIR1 0x04
#define RAWIR2 0x05
#define TA 0x06
#define TOBJ1 0x07
#define TOBJ2 0x08
// EEPROM
#define TOMAX 0x20
#define TOMIN 0x21
#define PWMCTRL 0x22
#define TARANGE 0x23
#define EMISS 0x24
#define CONFIG 0x25
#define ADDR 0x0E
#define ID1 0x3C
#define ID2 0x3D
#define ID3 0x3E
#define ID4 0x3F

  #define DHTPIN    32
  #define DHT2PIN   35                           
  #define DHT3PIN   34   
  #define RESETLINE 4
  #define LC_Clk    17 
  #define LC_Data   16 

  typedef struct
  {  unsigned long milcount;
    bool flag;
  } timer, *ptime;
  
  typedef struct 
  {  bool statusnow;
     bool statusold;
  } transferstate, *ptransferstate;
  
  typedef struct
  { transferstate OUT;
    transferstate ORIDE;
    bool sysflag;
  } heat, *pheat;
  
  typedef struct
  { transferstate OUT;
    transferstate IN;
    transferstate ORIDE;
    bool sysflag;
  } air, *pair;
  
  typedef struct
  { int oldform,newform; 
  }formx, *pformx;
  
  typedef struct  
  { formx FORM;
    int wifistate;
  } disp, *pdisp;
  
  typedef struct
  {  float temp;
     float humid;
  } dhtDAT, *pdhtDAT;
  
  typedef struct
  {  float temp;
    //float data[5];
     float average; 
     
  } tempstruct, *ptempstruct;
  typedef struct
  { bool goodrF;
    long calfactor;
    long mass;
  } loadc,*ploadc;
  
  typedef struct  
  { bool mode1,state;
    bool runningnow,done;
    bool startF[3]= {false,false,false};
    int heatP,cycleT,runT= 0;
    int maxT = 40;
    float massI,massG,massT = 0.0;
    int masspercent;
    unsigned long timer = 0.0;
    } dryp, *pdryp;
    
    typedef struct //opriginal code took these, *.02 (or /50) and then -273.some shit. 
   { uint16_t Object1; //store it this way, then print it with that operation
     uint16_t Object2;
     uint16_t Ambient;
     uint16_t Max;
     uint16_t Min;
   } irsens, *pirsens;
   
 typedef struct
  {float spectrum[6];
   float scale;
   irsens IR[4];
   dhtDAT INTAKE;
   dhtDAT EXHAUST;
   dhtDAT AMBIENT;
   bool spectrumF,loadcF,irF,dhtF;
  } sensors, *psensors;
  
  typedef struct
  { timer STATE[6];
    int currentstate; 
  } board, *pboard;
  
  typedef struct
  { board MCU;
    sensors SENSORS;    
    air AIR;
    dryp DRYP;
    heat HEAT; 
    disp DISP;
  } SYSstate, *pSYSstate;

AS726X sensor;  
HX711_ADC LC(LC_Data, LC_Clk);
SYSstate HDC;

void setup() 
 {  Wire.begin();
   Serial.begin(115200);
   pinMode(LC_Data,INPUT_PULLUP);
   pinMode(LC_Clk,OUTPUT); //CHECK THIS
    LC.begin();
    LC.start(1000);
    LC.setCalFactor(210.98);   
//INIT TIMERS
for(int k=0;k<4;k++)
    { HDC.MCU.STATE[k].milcount= millis();
      HDC.MCU.STATE[k].flag=false;
      yield();
    };

 };   
void loop() 
  { 
    if (millis() - HDC.MCU.STATE[0].milcount >=100L)
    { LC.update(); 
      HDC.MCU.STATE[0].milcount= millis();   
    };
    
  if (millis() - HDC.MCU.STATE[1].milcount >=1500L)
    { Serial.print("Load cell reading: ");
      HDC.SENSORS.scale=LC.getData();
      Serial.println(HDC.SENSORS.scale);
      HDC.MCU.STATE[1].milcount= millis(); 
    };       
    
 if (millis() - HDC.MCU.STATE[2].milcount >=5000L)
    {   readMLX();
        HDC.MCU.STATE[2].milcount= millis();     
     };
if (millis() - HDC.MCU.STATE[3].milcount >=10000L)
    { readASPEC();
     for (int j=0; j<6;j++)
      {
       Serial.print("Spec. bin "); Serial.print(j);Serial.print(" ");
       Serial.println(HDC.SENSORS.spectrum[j]); };
      HDC.MCU.STATE[3].milcount= millis();
     };    
  };

 void readMLX()
 {    //adafruit was reading these as floats then returning as double? maybe this fucked stuff up?
      HDC.SENSORS.IR[0].Ambient=readTemp(0x06); 
      HDC.SENSORS.IR[0].Object1=readTemp(0x07);
      HDC.SENSORS.IR[0].Object2=readTemp(0x08);
      HDC.SENSORS.IR[0].Max=readTemp(0x20); 
      Serial.print("stored value for mlx is ");
      Serial.println(HDC.SENSORS.IR[0].Object1);
      Serial.print("converted it is ");
      float temp = conv16(HDC.SENSORS.IR[0].Object1);
      Serial.println(temp);
      Serial.print("adafruits version of this was: ");
      Serial.println(double(temp));
   
 };   

 
  
 void readASPEC()
 {       HDC.SENSORS.spectrum[0]= sensor.getCalibratedR();
         HDC.SENSORS.spectrum[1]= sensor.getCalibratedS();
         HDC.SENSORS.spectrum[2]= sensor.getCalibratedT();
         HDC.SENSORS.spectrum[3]= sensor.getCalibratedU();
         HDC.SENSORS.spectrum[4]= sensor.getCalibratedV();
         HDC.SENSORS.spectrum[5]= sensor.getCalibratedW();
 };
uint16_t read16(uint8_t _addr,uint8_t a) 
{
  uint16_t ret;
  Wire.beginTransmission(a); // start transmission to device 
  Wire.write(a); // sends register address to read from
  Wire.endTransmission(false); // end transmission
  Wire.requestFrom(_addr, (uint8_t)3);// send data n-bytes read
  ret = Wire.read(); // receive DATA
  ret |= Wire.read() << 8; // receive DATA
  uint8_t pec = Wire.read();
  return ret;
};

uint16_t readTemp(uint8_t reg) 
{
 uint16_t temp; 
  temp = read16(MLX1AD,reg);
  return temp;
};

float conv16(uint16_t val) //use this to return the values in mlx data
{ float temp = val;       //originally outputted as a double
  temp *= .02;
  temp  -= 273.15;
  return temp;
};

```

### Debug Messages:
```

I can't enable core debug levels, there are no options for this board

00:00:19.469 -> Load cell reading: -0.13
00:00:20.953 -> Load cell reading: -0.13
00:00:21.470 -> stored value for mlx is 65535
00:00:21.470 -> converted it is 1037.55
00:00:21.470 -> adafruits version of this was: 1037.55
00:00:22.473 -> Load cell reading: -0.13
00:00:23.958 -> Load cell reading: -0.13
00:00:25.481 -> Load cell reading: -0.13
00:00:26.481 -> stored value for mlx is 65535
00:00:26.481 -> converted it is 1037.55
00:00:26.481 -> adafruits version of this was: 1037.55
00:00:26.481 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
00:00:26.481 -> Core 1 register dump:
00:00:26.481 -> PC      : 0x400ea924  PS      : 0x00060130  A0      : 0x800d12b0  A1      : 0x3ffb1ed0  
00:00:26.481 -> A2      : 0x00000000  A3      : 0x00000049  A4      : 0x3f401089  A5      : 0x3ffc12dc  
00:00:26.514 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000100  A9      : 0x00000001  
00:00:26.514 -> A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffb8274  A13     : 0x00000000  
00:00:26.514 -> A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
00:00:26.514 -> EXCVADDR: 0x00000126  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
00:00:26.548 -> 
00:00:26.548 -> Backtrace: 0x400ea924:0x3ffb1ed0 0x400d12ad:0x3ffb1ef0 0x400d1330:0x3ffb1f10 0x400d1394:0x3ffb1f30 0x400d13d9:0x3ffb1f50 0x400d0c98:0x3ffb1f70 0x400d0e85:0x3ffb1f90 0x400d3289:0x3ffb1fb0 0x40085aa1:0x3ffb1fd0
00:00:26.548 -> 
00:00:26.548 -> Rebooting...
00:00:26.548 -> ets Jun  8 2016 00:22:57
00:00:26.548 -> 
00:00:26.548 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:00:26.548 -> configsip: 0, SPIWP:0xee
00:00:26.548 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:00:26.582 -> mode:DIO, clock div:1
00:00:26.582 -> load:0x3fff0018,len:4
00:00:26.582 -> load:0x3fff001c,len:928
00:00:26.582 -> ho 0 tail 12 room 4
00:00:26.582 -> load:0x40078000,len:8424
00:00:26.582 -> ho 0 tail 12 room 4
00:00:26.582 -> load:0x40080400,len:5868
00:00:26.582 -> entry 0x4008069c
00:00:29.585 -> Load cell reading: -0.30

```

atanisoft, 2019-08-15:
@kbickham please decode the backtrace you have posted using the exception decoder in the issue template.

It would appear that you are attempting to access data from a null pointer. The decoded backtrace will point you to which code is causing it.

kbickham, 2019-08-15:
I'm not using any explicit interrupt and don't believe I could be hitting cap on ram.
```

Decoding stack results
0x400ea864: tcpip_adapter_sta_start at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c line 224
0x400d32d5: sendGData() at C:\Users\Lauren\Downloads\arduino nightly\sketches\newerwificonficpre2coretry/newerwificonficpre2coretry.ino line 721
0x40085aa1: psram_cache_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/spiram_psram.c line 832

```

kbickham, 2019-08-15:
Apologies, I actually tried to go through and clean up some some stuff in the program and that's what I got. The decoded backtrace for above is actually:

```
Decoding stack results
0x400ea924: tcpip_adapter_stop at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c line 261
0x400d3289: sendGData() at C:\Users\Lauren\Downloads\arduino nightly\sketches\newerwificonficpre2coretry/newerwificonficpre2coretry.ino line 721
0x40085aa1: psram_cache_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/spiram_psram.c line 832
```

From what I read..the ram usage is an issue for a lot of people? This version of my code isn't even utilizing any internet protocol of any kind, I was just reading from 3 sensors (mlx90614 ir sensors, as7263 spectrometer, and a loadcell through an hx711 amplifier/adc)

atanisoft, 2019-08-15:
@kbickham the decoded stack doesn't match any of the code you have provided in this issue. It appears that sendGData() is attempting to use a null object in it's call to tcpip_adapter_stop().

kbickham, 2019-08-15:
Changed my ide version/reinstalled tools and boards. May not be accurate...I have 24 hours to finish a project, I'll repost pertinent info asap.

kbickham, 2019-08-17:
Pretty sure it was the dev version of wifimanager. I cut down the mem use and switched to only using a captive portal.

closed 2019-08-22 by kbickham