espressif/arduino-esp32 #3094: Any more releases?

judge2005, 2019-08-16:
Hi

Just wondering when the next release will be coming out. It has been a while.

atanisoft, 2019-08-16:
There is a 1.0.3-rc currently, the last I heard was that @me-no-dev was waiting for an update on the esp-idf side for 3.2.x release. I know @me-no-dev has been quite busy the last couple weeks but I'm sure he will get a new rc out soon if not a final 1.0.3 release.

axa88, 2019-08-23:
A related question. How are updates to the esp idf included?
Are all updates included in new builds or they are chosen on what basis?
I ask as I am looking for a particular fix [bug fix](https://github.com/espressif/esp32-bt-lib/commit/717f0c6ec71a016a0b292acffeacf239d007b8ff) but didn't notice any mention of it in recent [release candidate notes](https://github.com/espressif/arduino-esp32/releases).

lbernstone, 2019-08-23:
The timing is somewhat arbitrary, when me-no-dev has time to match the changes in idf to the codebase here.  The version chosen will always be a stable version (that is currently 3.2).  He grabs the latest code from that branch and works through the various changes until it integrates cleanly, then makes a release.  If you just want to know what it is currently based off, [search the commits for update idf](https://github.com/espressif/arduino-esp32/search?q=update+idf&type=Commits)

asetyde, 2019-08-28:
I'm confused, I see 
[image]

asetyde, 2019-08-28:
Ok but with many wifi bugs that I see in this last months ... stable is a heavy word. We use more than 3000 ESP32 but it's difficult if base of firmware is so volatile and bugged ... it's my opinion but at now we believe in this project

asetyde, 2019-08-28:
Yes you ve right, but I use 1.0.2, but sometimes with use of i2c and SPI many of this WiFi freeze, with 1.0.3x Sometimes reboot but no more freeze ... 
Old version has many bugs on Arduino Library side.
I hope on 1.0.3 stable e we remain on this for many times

closed 2019-11-10 by stale[bot]