espressif/arduino-esp32 #3097: ESP32 restarts on calling WiFi.softap(ssid,pw);

zekageri, 2019-08-16:
ESP32 restarts when calling WiFi.softap(); Sometimes it is working, sometimes it is not.
Debug verbose does not show anything. My esp just restarts itself instantly when i call softap.
My power supply is a decent one. It is an EMG18144 big ass adjustable Supply.
I'am using the latest arduino core, on PlatformIO in vscode.

**_What i tried:_**

With the commented out things combinations too.

static const inline void WiFiOn(){
    WiFi.disconnect(true);
    delay(100);
    WiFi.enableAP(true); 
    delay(100);
    //WiFi.softAP(ssid,password, 10, 0, 1);
    WiFi.softAP(ssid,password);
    delay(150);
    //WiFi.mode(WIFI_AP);
}

paulcool234, 2019-08-21:
Do you use a development board? If not, add a cap near the esp32 voltage input pins.
Sudden restarts are brownouts most of the time.

zekageri, 2019-08-21:
> Do you use a development board? If not, add a cap near the esp32 voltage input pins.
> Sudden restarts are brownouts most of the time.

I have a bare ESP32-WROOM-32 chip. I have capacitors everywhere on the circuit. I got one on the supply pins, one on a tft display supply pin and one more on the esp32 supply pin. :/

closed 2019-08-26 by zekageri