espressif/arduino-esp32 #3098: How to build everything from source instead of static libraries?

piotrkundu, 2019-08-16:
This project is based on pre-built static libraries, which means you can't make any changes.
I would like to change the UUID for BT SPP and use it in Arduino https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/esp_spp_api.c#L26

 How?

atanisoft, 2019-08-16:
Short answer... Not possible in Arduino IDE...

Longer answer.. You have two options:
1) Use Arduino as an IDF component, documented [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).
2) Use PlatformIO IDE with the development version of the Espressif32 platform using the framework setting as "espidf, arduino" and having the sdkconfig.h in your src directory.

Option 2 is a lot easier to setup and maintain over the long term. As an added bonus you will get a full fledged IDE.

lbernstone, 2019-08-16:
[option 2](https://github.com/platformio/platform-espressif32#usage)

atanisoft, 2019-08-16:
for option 3, make sure to use the correct version of esp-idf... currently set to the 3.2 release branch..

piotrkundu, 2019-08-21:
I tried to build libbt.a in CLion and then copy them to ~/Library/Arduino15/packages/esp32 folder structure. Since my esp-idf is latest master (v4) and the Arduino libs are v3.x I got a lot of missing symbols so I had to build and copy another 3 libs. Of course once I got a ELF file built and flashed it didn't work, so I kind of dropped that for now.

@atanisoft Arduino is great for hobbyists or some rapid prototyping. I was just surprised that the binaries (libs) are checked into git in arduino-esp32 and the folder structure is flattened to just sdk/lib. When you build these libraries with cmake, idf.py or CLion the libs are build into a folder structure of esp-idf/component_name/libxxx.a. I guess I could copy *.a from esp-idf to ~/Library/Arduino15/packages/esp32

For Option 2, I've actually installed Visual Studio Code, PlatformIO and ESP32 component. That's a good free option as well.

atanisoft, 2019-08-21:
Since you have CLion, I'd suggest use the PIO plugin for it and you can follow the same options for #2.

piotrkundu, 2019-09-11:
Got everything I need up and running in CLion - no luck with Arduino IDE.

closed 2019-09-11 by piotrkundu

RdlP, 2020-07-18:
I am facing a similar problem. I want to re-compile libbt and replace it in the arduino library folder. But my problem is how I can build the libbt.a library?