espressif/arduino-esp32 #3099: HSPI frequency w/ SD card maxed out at 16MHz

christienmiller, 2019-08-16:
### Hardware:
Board:							TTGO T-Display ESP32 Schematic: [ESP32-TFT(6-26).pdf](https://github.com/espressif/arduino-esp32/files/3511255/ESP32-TFT.6-26.pdf)
Core Installation version:			latest commit fa55a2c91d31eab673c0115bc57e9437350488cc
IDE name:						Arduino IDE
Flash Frequency:					240Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
When using HSPI and a SD card I cannot get the frequency of to go past 16MHz - setting to 18MHz seem to have no effect when looking at SPCLK on the scope and setting to 20MHz and higher leads to no SD card at all.  Also, this board has a small SPI display which also appears to be using HSPI according to TFT_eSPI.cpp - no clue how I can have two conflicting SCLK, MISO, and MOSI pin configs - both the display and the SD card are working..

TFT_eSPI library found here: https://github.com/Bodmer/TFT_eSPI

I am trying to get better read performance and setting the clock to 20MHz or higher should help.  According to ESP32 docs (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html) I should be able to get at least 26.6MHz on HSPI.  Is there a cap when using Arduino with the ESP32?

### Sketch:
```cpp
#include <TFT_eSPI.h>
#include <SPI.h>
#include <SD.h>
#include "bmp.h"

#define SD_CS      15
#define SD_SCK     2 //14 PWN_EN pin
#define SD_MOSI    13
#define SD_MISO    12

#ifndef TFT_DISPOFF
#define TFT_DISPOFF 0x28
#endif

#ifndef TFT_SLPIN
#define TFT_SLPIN   0x10
#endif

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

TFT_eSPI tft = TFT_eSPI(135, 240); // Invoke custom library
SPIClass SDSPI(HSPI);

unsigned long startTime;
unsigned long timeToRead;
unsigned long timeToPushImage;
char buff[512];
int vref = 1100;
int btnCick = false;
uint16_t picture[128][240];
// TO CONVERT VIDEO TO RAW: ffmpeg -i drac_sample.mp4 -vf scale=240:128 -c:v rawvideo -pix_fmt rgb565le drac.rgb

void setup()
{
    SDSPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
    SD.begin(SD_CS, SDSPI, 16000000);
    
    Serial.begin(115200);
    
    delay(1000);
    
    Serial.println("Start");
    Serial.println("SPI frequency at " + String(SDSPI.getFrequency()));

    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    tft.init();
    tft.setRotation(1); // landscape
    tft.fillScreen(TFT_BLACK);
    tft.setTextSize(2);
    tft.setTextColor(TFT_WHITE);
    tft.setCursor(0, 0);
    tft.setTextDatum(MC_DATUM);
    tft.setTextSize(1);

    if (TFT_BL > 0) {
        pinMode(TFT_BL, OUTPUT);
        digitalWrite(TFT_BL, HIGH);
    }
    tft.setSwapBytes(true);
    tft.pushImage(0, 0,  240, 135, ttgo);
    

    list_files();
    
    delay(5000);

    int i = 1;
    while (i--) {
        tft.fillScreen(TFT_RED);
        delay(1000);
        tft.fillScreen(TFT_BLUE);
        delay(1000);
        tft.fillScreen(TFT_GREEN);
        delay(1000);
    }
}

void list_files()
{
   tft.setCursor(0, 0);
   static File file;
      Serial.println("list");
      File root = SD.open("/");
      file = root.openNextFile();
      while (file) {
        if (file.isDirectory()) {
          Serial.print("DIR: ");
          Serial.println(file.name());
          sprintf(buff,
            file.name());
          tft.println(buff);
        }
        else {
          Serial.print("FILE: ");
          Serial.print(file.name());
          Serial.print("  SIZE: ");
          Serial.println(file.size());
          sprintf(buff,
            file.name());
          tft.println(buff);
        }
        file = root.openNextFile();
      }
      Serial.println();
}

void play() {
  static File file;
  file = SD.open("/drac.rgb");
  tft.fillScreen(TFT_BLACK);
  Serial.println("file playback..");
  while(file.available()) {
    startTime = millis();
    file.read((uint8_t*)picture, 240*128*2);
    timeToRead = millis();
    tft.pushImage(0, 3, 240, 128, (uint16_t*)picture);
    timeToPushImage = millis();
    Serial.print("Read: ");
    Serial.print(timeToRead - startTime); //prints time passed since last frame
    Serial.print("ms Push: ");
    Serial.print(timeToPushImage - timeToRead); //prints time passed since last frame
    Serial.print("ms FPS: ");
    Serial.println( String(1000.0 / (timeToPushImage - startTime)) ); //prints time passed since last frame
  }
  file.close();
  Serial.println("file done");
}

void loop()
{
   play();
}

```

christienmiller, 2019-08-17:
Actually it does appear to be trying to run at higher speeds, if I set it to 40MHz I can see it try to open the SD card at that speed but fails to detect the card.

atanisoft, 2019-08-17:
Have you tried sdmmc instead of SD?

christienmiller, 2019-08-17:
I have not yet - been looking at it but looks like the pins used cannot be re-mapped.  Two of the GPIOS (14 and 4) are being used for other things on the T-Display board.  I can do some rework on the board to free up those inputs I think.   Does 4-bit mode really improve performance that much?  I would hope close to 4x the amount..

atanisoft, 2019-08-18:
I'm not certain.

christienmiller, 2019-08-20:
Had a chance to try SDMMC and it is a little faster.  I can get about 1MB/s now which ends up being 13-14fps on the display - not bad.  I scoped the SCLK and it is running at 40MHz.  I have also noted that the SDMMC mode is pretty darn picky and only 1 out of my 3 micro SD cards work - one class 4.  Class 3 and 10 not working.  I wouldn't be surprised if this is due to using a breadboard with long leads..

lbernstone, 2019-08-20:
It is also picky about the format of the cards.  If you let it reformat them, they will likely work in sdmmc.
If not, check the impedance of the pullup resistors. https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#hardware

closed 2019-11-02 by stale[bot]

JaimeAlbq, 2020-01-18:
Hey @christienmiller!

Are you still having problems using HSPI?

I was facing this problem just when I need to use the RESET button. It went to a infinity boot.

So to solve this I discovered that the SD card needs to be turn off enough time to the main boot of the ESP32 starts.

To delay this time I used the logical MOSFET [IRLI2203N](http://www.irf.com/product-info/datasheets/data/irli2203n.pdf) cutting the connection throw the ground of my SD module by driving from the RESET button from the board.

[image]
[image]