espressif/arduino-esp32 #3100: changing configUSE_PREEMPTION=0 in FreeRTOSConfig.h changes nothing

AminAbdiyan, 2019-08-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
I am new in the field of FreeRTOS, I change configUSE_PREEMPTION=0 in FreeRTOSConfig.h file but changes nothing.
I only use two simple tasks, I expected that was run only one task but it was run both of them.

###Code 
void Task1(void *pvParameters)  // This is a task.
{
  (void) pvParameters;
  for (;;) // A Task shall never return or exit.
  {
    Serial.println("Hello from Task1");
  }
}

void Task2(void *pvParameters)  // This is a task.
{
  (void) pvParameters;
  for (;;)
  {
    Serial.println("Hello from Task2");
    Serial.println(configUSE_PREEMPTION);
  }
}
### Serial output
Hello from Task2
0
Hello from Task2Hello from Task1
Hello from Task1
0
Hello from Task2
0
Hello from Task2

atanisoft, 2019-08-17:
Changing the sdkconfig.h file will not alter the behavior unless the FreeRTOS module of esp-idf is recompiled. In the Arduino IDE this is provided as a pre-compiled binary only. The only solution to what you are wanting to do by disabling preempting is Arduino as an IDF component.

AminAbdiyan, 2019-08-17:
> Changing the sdkconfig.h file will not alter the behavior unless the FreeRTOS module of esp-idf is recompiled. In the Arduino IDE this is provided as a pre-compiled binary only. The only solution to what you are wanting to do by disabling preempting is Arduino as an IDF component.

thank you for your reply, can you explain how can I do it.

atanisoft, 2019-08-17:
1. Use Arduino as an IDF component, documented [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).
2. Use PlatformIO IDE with the development version of the Espressif32 platform using the framework setting as "espidf, arduino" and having the sdkconfig.h in your src directory. As documented [here](https://github.com/platformio/platform-espressif32#usage).
3. Copy all relevant files from [esp-idf](https://github.com/espressif/esp-idf/tree/release/v3.2) to your sketch folder and it should override the static library content.

I'd highly suggest using option 2 as it will be the easiest to setup and there are many other benefits of PlatformIO IDE over Arduino IDE.

AminAbdiyan, 2019-08-21:
> 1. Use Arduino as an IDF component, documented [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).
> 2. Use PlatformIO IDE with the [development/upstream version](https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream) of the Espressif32 platform using the framework setting as "espidf, arduino" and having the sdkconfig.h in your src directory. As documented [here](https://github.com/platformio/platform-espressif32#usage).
> 3. Copy all relevant files from [esp-idf](https://github.com/espressif/esp-idf/tree/release/v3.2) to your sketch folder and it should override the static library content.
> 
> I'd highly suggest using option 2 as it will be the easiest to setup and there are many other benefits of PlatformIO IDE over Arduino IDE.

thank you for your response. but still, I couldn't do that.

first I tried with method 2 but there is not configUSE_PREEMPTION in sdkconfig.h, I assign configUSE_PREEMPTION=0 in FreeRTOSConfig.h file and compile it with PlatformIO IDE, but result still was the same.

I used 1 method, I assign configUSE_PREEMPTION=0 and run make and after make I replace created libfreertos.a in Arduino libraries and compile it. but again the result was the same.

Would you mind helping me more?

lbernstone, 2019-08-21:
This is not a sdkconfig.h option.  You will need to change the assignment in FreeRTOSConfig.h and compile completely from source.  Since you say you have done that, and it is not working, there is not much else we can do to help you.  If you copy the libfreertos.a into Arduino, make sure you also then change the FreeRTOSConfig.h in the Arduino includes.

AminAbdiyan, 2019-08-21:
> This is not a sdkconfig.h option. You will need to change the assignment in FreeRTOSConfig.h and compile completely from source. Since you say you have done that, and it is not working, there is not [...]

> This is not a sdkconfig.h option. You will need to change the assignment in FreeRTOSConfig.h and compile completely from source. Since you say you have done that, and it is not working, there is not [...]

I did but no chance!

atanisoft, 2019-08-21:
Perhaps what you are trying to do is not supported. The FreeRTOS kernel is preemptive in nature by default and there may not be a definitive way to disable it. You *CAN* however use a semaphore/mutex passed between your tasks if you require such locking to ensure only one of your tasks runs at a time. But keep in mind there are other tasks which *WILL* run regardless of your tasks and semaphore/mutex.

closed 2019-11-03 by stale[bot]