espressif/arduino-esp32 #3109: Make Issue in strings.h

wjundong, 2019-08-18:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                   yes
Upload Speed:					115200
Computer OS:                                        Ubuntu
when I make my project, the Issue is
/home/shino/esp/Project/Arduino/components/arduino/libraries/AzureIoT/src/az_iot/c-utility/inc/azure_c_shared_utility/strings.h:18:1: error: unknown type name 'IF'
 MOCKABLE_FUNCTION(, STRING_HANDLE, STRING_new);
 ^~~~~~~~~~~~~~~~~

## in main.cpp
#include "Arduino.h"
extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);
    digitalWrite(4, HIGH);
    //do your own thing
}

wjundong, 2019-08-18:
Work tree
├── CMakeLists.txt
├── components
│   └── arduino
├── example_test.py
├── main
│   ├── CMakeLists.txt
│   ├── component.mk
│   ├── Kconfig.projbuild
│   └── main.cpp
├── Makefile
├── README.md
├── sdkconfig
├── sdkconfig.defaults
└── sdkconfig.old

wjundong, 2019-08-18:
How can I solve this error prompted by the compiler?

jubueche, 2019-08-26:
I have the same problem.

closed 2019-11-08 by stale[bot]