espressif/arduino-esp32 #3113: 烧录固件后系统一直重启

ZorkHuang, 2019-08-19:
### Hardware:
Board:							ESP32-WROOM-32U
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10
Flash Size:                                               2MB

### Description:
烧录固件后串口一直重启，硬件外围电路自己开发的
GPIO0：BOOT1启动项
GPIO2：下拉地
GPIO15：上拉3.3v

### Sketch: 
uint32_t print_tick = 0;
uint32_t count = 0;
#define IO_PWR  14
#define IO_COM  33
#define IO_K4	13

void setup()
{
	pinMode(IO_PWR, OUTPUT);
	pinMode(IO_COM, OUTPUT);
	pinMode(IO_K4, OUTPUT);
	digitalWrite(IO_PWR, HIGH);
	digitalWrite(IO_COM, LOW);
	digitalWrite(IO_K4, LOW);
	Serial.begin(115200);
	Serial.println();
	Serial.println();
	Serial.println("Contactor v1.1-2019/08/19");
}

void loop()
{
	uint32_t now = millis();
	if (now >= print_tick) {
		Serial.print("count:");
		Serial.print(count++);
		Serial.print(",time:");
		Serial.print(now);
		Serial.println(" ms");
		print_tick += 500;
		digitalWrite(IO_PWR, count % 2);
	}
}

### Debug Messages:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
E (102) cpu_start: Running on single core chip, but application is built with dual core support.
E (102) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.
abort() was called at PC 0x40082b66 on core 0

Backtrace: 0x40088dc4:0x3ffe3be0 0x40088ff1:0x3ffe3c00 0x40082b66:0x3ffe3c20 0x40078f2b:0x3ffe3c40 0x40078f91:0x3ffe3c70 0x40078f9c:0x3ffe3ca0 0x40079165:0x3ffe3cc0 0x400806da:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

lbernstone, 2019-08-20:
https://github.com/lbernstone/arduino-esp32-solo
v1.0.2 libraries

ZorkHuang, 2019-09-27:
> https://github.com/lbernstone/arduino-esp32-solo
> v1.0.2 libraries

Thanks

closed 2019-12-10 by stale[bot]