espressif/arduino-esp32 #3116: ESP32 mbedtls_internal_sha512_process exception

nejcmedved, 2019-08-20:
I am having an issue with TLS communication. I have set cert and private key as shown on tutorial

client.setCertificate(test_client_cert);
client.setPrivateKey(test_client_key);

after I try to do connect exception like shown below pops up.

```
PC: 0x401050fa: mbedtls_internal_sha512_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/sha512.c line 242
EXCVADDR: 0x00000000

Decoding stack results
0x401050fa: mbedtls_internal_sha512_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/sha512.c line 242
0x40105ff4: mbedtls_sha512_update_ret at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/sha512.c line 322
0x4010658e: mbedtls_sha512_ret at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/sha512.c line 435
0x400fd931: entropy_update at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/entropy.c line 192
0x400fda1e: entropy_gather_internal at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/entropy.c line 288
0x400fdb08: mbedtls_entropy_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/entropy.c line 363
0x400fd6b2: mbedtls_ctr_drbg_reseed at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/ctr_drbg.c line 337
0x400fd79b: mbedtls_ctr_drbg_seed_entropy_len at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/ctr_drbg.c line 98
0x400fd7d6: mbedtls_ctr_drbg_seed at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/mbedtls/library/ctr_drbg.c line 111
0x400e0db6: start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*) at C:\Users\Nejc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFiClientSecure\src\ssl_client.cpp line 94
0x400e09b9: WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at C:\Users\Nejc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFiClientSecure\src\WiFiClientSecure.cpp line 129
0x400e0ad5: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Nejc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFiClientSecure\src\WiFiClientSecure.cpp line 111
0x400d6755: CloudConnector::TaskFunction(void*) at C:\Users\Nejc\AppData\Local\Temp\arduino_build_190776\sketch/CloudConnector.h line 166
0x40088ff5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

Any ideas what could be wrong?

nejcmedved, 2019-08-20:
Here is also debug output.

```
[V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 250124
[V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[V][ssl_client.cpp:90] start_ssl_client(): Seeding the random number generator
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (CloudConnector) 
```

atanisoft, 2019-08-20:
@nejcmedved This output says that your task "CloudConnector" is running out of stack space.

nejcmedved, 2019-08-20:
Thank you very much. Will try to fix it. As soon as I do that I will post here and we can close it.

Best, Nejc

closed 2019-11-02 by stale[bot]