espressif/arduino-esp32 #3122: Esp32 continuously reset in a loop due to that board is not working

Embel-Technologies, 2019-08-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz and 80Mhz
PSRAM enabled:                                     yes
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
While uploading a WiFi and MQTT code on self developed esp-32 board, following errors or outcome occurs due to that program and board is not working as expected. It continuously rst:0x7 (TG0WDT_SYS_RESET) reset the board. Please give me solution on this I am stuck on this issue and need to solve it asap.  

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>        // Include the Wi-Fi library
 
const char* ssid     = "****";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "****";     // The password of the Wi-Fi network
 
void setup() {
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');
  
  WiFi.begin(ssid, password);             // Connect to the network
  Serial.print("Connecting to ");
  Serial.print(ssid);
 
  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(500);
    Serial.print('.');
  }
  Serial.println('\n');
  Serial.println("Connection established!");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());         // Send the IP address of the ESP32 to the computer
} 
void loop() {  
}

### Debug Messages:
```
 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 -> configsip: 0, SPIWP:0xee
 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 -> mode:DIO, clock div:1
 -> load:0x3fff0018,len:4
 -> load:0x3fff001c,len:1100
 -> load:0x40078000,len:9232
 -> load:0x40080400,len:6400
 -> entry 0x400806a8
 -> ets Jun  8 2016 00:22:57
 -> 
 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 -> configsip: 0, SPIWP:0xee
 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 -> mode:DIO, clock div:1
 -> load:0x3fff0018,len:4
-> load:0xbfff009c,len:1484
 -> load:0x4000c44c,len:-1073807348
 -> 1162 mmu set 00010000, pos 00010000
 -> 1162 mmu set 00020000, pos 00020000
 -> 1162 mmu set 00030000, pos 00030000
 -> 1162 mmu set 00040000, pos 00040000
 -> 1162 mmu set 00050000, pos 00050000
 -> 1162 mmu set 00060000, pos 00060000
 -> 1162 mmu set 00070000, pos 00070000
 -> 1162 mmu set 00080000, pos 00080000
 -> 1162 mmu set 00090000, pos 00090000
 -> 1162 mmu set 000a0000, pos 000a0000
 -> 1162 mmu set 000b0000, pos 000b0000
 -> ho 0 tail 12 room 4
 -> load:0xd595baa3,len:-121089839
 -> 1162 mmu set 000c0000, pos 000c0000
 -> 1162 mmu set 000d0000, pos 000d0000
 -> 1162 mmu set 000e0000, pos 000e0000
 -> 1162 mmu set 000f0000, pos 000f0000
 -> 1162 mmu set 00100000, pos 00100000
 -> 1162 mmu set 00110000, pos 00110000
 -> 1162 mmu set 00120000, pos 00120000
 -> 1162 mmu set 00130000, pos 00130000
 -> 1162 mmu set 00140000, pos 00140000
 -> 1162 mmu set 00150000, pos 00150000
 -> 1162 mmu set 00160000, pos 00160000
 -> 1162 mmu set 00170000, pos 00170000
 -> 1162 mmu set 00180000, pos 00180000
 -> 1162 mmu set 00190000, pos 00190000
 -> 1162 mmu set 001a0000, pos 001a0000
 -> ets Jun  8 2016 00:22:57
 -> 
-> rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 -> configsip: 0, SPIWP:0xee
 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 -> mode:DIO, clock div:1
 -> load:0x3fff0018,len:4
 -> load:0xbfff009c,len:1228
 -> load:0x3f720080,len:1073782456
 -> ets Jun  8 2016 00:22:57
 -> 
-> rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 -> configsip: 0, SPIWP:0xee
 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 -> mode:DIO, clock div:1
 -> load:0x3fff0018,len:4
 -> load:0xbfff009c,len:1228
 -> load:0x3f720080,len:1073782456
 -> ets Jun  8 2016 00:22:57
 -> 
 -> rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 -> configsip: 0, SPIWP:0xee
 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 -> mode:DIO, clock div:1
 -> load:0x3fff0018,len:4
 -> load:0xbfff009c,len:1228
 -> load:0x3f720080,len:1073782456
 -> ets Jun  8 2016 00:22:57

```

paulcool234, 2019-08-21:
Do you have the same problem if you use the default blink sketch?

Embel-Technologies, 2019-08-21:
Yes sometimes but blink sketch is work properly

lbernstone, 2019-08-21:
Try to [erase the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region)
Then upload the code again.  If it continues to fail, then you have a power issue.  Try another USB cable or a different USB port.

Embel-Technologies, 2019-08-21:
I have done that but sketch is not working

kapyaar, 2019-08-22:
Can you post a screenshot of what is the settings that you use to program?

Embel-Technologies, 2019-08-23:
[image]

JAntHarrison, 2019-08-29:
Does your board use an external crystal or the internal clock? An inconsistent clock signal could be the reason behind the resets.

A much more likely case would be a power issue. Check if your power source is able to deliver enough current. (The fact that your board functions well with blink but resets with a more demanding code could be due to power requirements, especially since your MQTT seem to use Wifi - make sure to be able to supply 500mA+ current during bootup)

closed 2019-11-11 by stale[bot]

VeneKone, 2023-11-02:
Whenever this happens, it is almost always a power supply issue - at least this was the case with my project. I was using a NodeMCU ESP32S chip along with a 12V adapter connected to a 3-pin voltage regulator which gave 5V supply to the `5V` pin of my ESP32.
I could see that the red on-board LED (which turns on whenever there's a supply) was twinkling like a distant star, indicating some sort of fluctuation in the input supply. It's mostly because of loose wire-connections, which can happen on breadboards, but in my case it was due to the 12V adapter. It was noisy.

The moment I switched to a 12V Li ion battery, which provided a stable supply, the twinkling stopped and my chip was no longer rebooting on its own.

Reboot loops can also occur due to heavy/time-consuming work being done in an interrupt-service-routine, but that usually yields a "Guru Meditation" error.