espressif/arduino-esp32 #3123: collectHeaders doesn't support repeated names

lgg2, 2019-08-22:
Hello,

With a M5Stack grey (ESP32)

I am trying to recovery the cookies from a server that send three lines of "Set-Cookie"

> Set-Cookie: LoginError=0; Expires=Thu, 01 Jan 1970 01:00:00 CET; Max-Age=0; Path=/; Version="1"
Set-Cookie: setInitVars=1; Expires=Tue, 19 Jan 2038 04:13:43 CET; Max-Age=10; Path=/; Version="1"
Set-Cookie: pksession=11670; Expires=Tue, 19 Jan 2038 04:13:43 CET; Max-Age=600; Path=/; Version="1"

But `collectHeaders` only saves the last one.

``` C/C++
HTTPClient http;
.
.
const char *hdrs[] = {"Set-Cookie","Set-Cookie","Set-Cookie"};
size_t numhdrs = sizeof(hdrs)/sizeof(char*);
.
.
http.collectHeaders(hdrs, numhdrs);
.
.
printf("%s\n%s\n%s\n", http.header((size_t) 0).c_str()), http.header((size_t) 1).c_str()), http.header((size_t) 2).c_str()));
```
Prints only
```
Set-Cookie: pksession=11670; Expires=Tue, 19 Jan 2038 04:13:43 CET; Max-Age=600; Path=/; Version="1"
```

lbernstone, 2019-08-22:
I assume you have your parens correct in your actual code, but just to uncomplicate things, try
`printf("%s\n%s\n%s\n", http.header(0U), http.header(1U), http.header(2U));`

Jeroen88, 2019-08-23:
You should use:

`const char *hdrs[] = {"Set-Cookie"};
`

Each header with the same name will be apended in header("Set-Cookie") seprated by a comma. In your code header(1) and header(2) are not present and will return an empty String(). I would expect though that header(0) contained all three headers. Are you sure the server sends them?

closed 2019-11-05 by stale[bot]