espressif/arduino-esp32 #3127: Rebooting...(xQueueGenericSendFromISR)- assert failed! abort() was called at PC 0x400896e0 on core 1

SalmaG98, 2019-08-23:
### Hardware:
Board:							NodeMCU-32S
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					912600
Computer OS:                                         Ubuntu

### Description:
I am working on a connected collar for pets that sends body temperature, pulse and position in JSON format over MQTT to an HTTP server for it to be processed, I started by testing the "random" Data_Sending code which ran perfectly well, then I started working on the pulse sensor SEN-11574 I found a code that works with timer interrupts and tweaked it a bit and it gave accurate values. As soon as I started merging and multiprocessing both codes I started getting a lot of errors.
[image]
[image]

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//LIBRARIES
#include <NTPClient.h>
#include <WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>
//#include <queue.h>

//VARIABLES
//UDP_NTP
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);
String Collar_History;

//task

TaskHandle_t Sending_Data_task;
boolean once = true;

//Connecting

const char* ssid = "ODCGuest";
const char* password =  "T5L4zPovSG";
const char* mqttServer = "broker.hivemq.com";
const int mqttPort = 1883;
const char* clientID = "24727054";
const char* channelName = "oprex/collarHistory";

//BPM

#define pulsePin 36

// these variables are volatile because they are used during the interrupt service routine!
 int BPM;                   // used to hold the pulse rate
 int Signal;                // holds the incoming raw data
 int IBI = 0;             // holds the time between beats, must be seeded! 
 boolean Pulse = false;     // true when pulse wave is high, false when it's low
 boolean QS = false;        // becomes true when Arduoino finds a beat.

 int rate[10];                    // array to hold last ten IBI values
 unsigned long sampleCounter = 0; // used to determine pulse timing
 unsigned long lastBeatTime = 0;  // used to find IBI
 int P =512;                      // used to find peak in pulse wave, seeded
 int T = 512;                     // used to find trough in pulse wave, seeded
 int thresh = 512;                // used to find instant moment of heart beat, seeded
 int amp = 100;                   // used to hold amplitude of pulse waveform, seeded
 boolean firstBeat = true;        // used to seed rate array so we startup with reasonable BPM
 boolean secondBeat = false;      // used to seed rate array so we startup with reasonable BPM
 int rx_Signal;
 portBASE_TYPE xHigherPriorityTaskWokenByPost;

hw_timer_t * timer = NULL;

xQueueHandle Sig_Queue = 0;
[... 219 lines cut ...]
    pinMode(pulsePin,INPUT_PULLDOWN);
    Serial.begin(9600);
    xTaskCreatePinnedToCore(Sending_Data,"Data_Sending",2500,NULL,2,&Sending_Data_task,0);
    delay(500);
    interruptSetup();   
  }

void loop() 
  {
    
  //sendDataToProcessing('S', Signal);     // send Processing the raw Pulse Sensor data
    if (QS == true){                       // Quantified Self flag is true when arduino finds a heartbeat
          Serial.println("BPM: "+ String(BPM));   // send heart rate with a 'B' prefix
          //sendDataToProcessing('Q',IBI);   // send time between beats with a 'Q' prefix
          QS = false;  
       }

    //delay(1000);
    yield();
    
  }
### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

atanisoft, 2019-08-23:
Sig_Queue is null on the first time the Timer is triggered. You should set this up in setup() directly before calling xTaskCreatePinnedToCore so that all usages will have the queue available to use.

atanisoft, 2019-08-23:
this is not a bug in arduino-esp32 or esp-idf...

stickbreaker, 2019-08-23:
@SalmaG98 The `Assert` error you are seeing is caused by using `xQueueSendFromISR()` before the Queue is created.

You create the queue inside your `Sending_data()` task, after you connect WiFi. The WiFi connection time is variable, and may never happen.  You create the `Sending_data()` task in `setup()` then wait 500ms then attach the timer ISR and *ASSUME* the queue has been create and is ready.

Since `Sending_data()` is controlling when/if the Queue is create, have it attach your `timerISR()` after the Queue exists.

Chuck.

SalmaG98, 2019-08-23:
@atanisoft I'm sorry I didn't know in which repo to create this issue

atanisoft, 2019-08-23:
@SalmaG98 none, it is a bug in your code. You are using the queue before you have created it!

SalmaG98, 2019-08-23:
Sorry it's my first time working on ESP32 or multiprocessing, I've changed the setup() to this

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup() 
  {
      
    pinMode(pulsePin,INPUT_PULLDOWN);
    Serial.begin(9600);
    Sig_Queue = xQueueCreate(10, sizeof(int));
    interruptSetup(); 
    xTaskCreatePinnedToCore(Sending_Data,"Data_Sending",8192,NULL,2,&Sending_Data_task,0);
    delay(500);  
  }

### Debug Messages:
```
but now I get this and I get nothing by pasting it in the ESP Exception Decoder:

[image]

atanisoft, 2019-08-23:
That indicates you are trying to do something inside the timer ISR callback which is not supported. The first thing I can see is usage of printf() inside an ISR, that is definitely going to cause uses. Switch to ets_printf instead as a temporary solution (it would be better to get rid of printf entirely in the ISR!)

Try simplifying your ISR usage as there are likely other issues in it.

SalmaG98, 2019-08-24:
The ISR is pretty short, it used to comprehend the whole Get_BPM method but I simplified the ISR and sent the raw signal in the queue for it to be processed separately I have no clue on how to further simplify it, even after commenting the printf part I still get the same error I don't know if it's a RAM thing or something else.
Any tips on how to simplify the ISR? Thank you!

atanisoft, 2019-08-24:
About the only thing I can think of is remove any external variable references from the ISR code. Anything that is outside the ISR should be considered read-only or unavailable (unless you explicitly tag is for ISR access, there are restrictions on use for this). I'd also suggest sending the raw data to the queue and let the consumer handle the "map" call as well since that function is not tagged as ISR safe.

closed 2019-11-06 by stale[bot]