espressif/arduino-esp32 #3128: read mid line in sd card

mehrdad987, 2019-08-23:
hi, 
I don't know if I can read and write different lines with SD.h 
I wanna read from a specific line in my file and not all the data in my file
but I couldn't create it
s.read(line[0:4])
is it possible?

stickbreaker, 2019-08-23:
Simple answer no.

as an example:  Pickup any book you have,  Read only the 15th paragraph.  Do not Read the first 14. Jump exactly to the 15th paragraph.  I hope you can see this is an impossible problem.  

Files have two modes of access, 
* Sequential: start at beginning read until the end.
* Random: Jump to a specific BYTE offset, read a specific number of bytes, or to a specific termination character.

Too get to the the 5th line(as demarcated  by a `newline` character) you have to read each line and count the `newline` characters.  If you knew that each line is composed of EXACTLY 'n' character you could jump to n*(5-1).

```c++
SD.begin()
File f = SD.open("myfile",FILE_READ);
int lineLen = 25;
int fileLength = f.size();
int offset = lineLen *(4); // starting position of fifth line
#define BUFFMAX 512
int buffMax = BUFFMAX;
char buffer[BUFFMAX+1]; //extra byte for null terminator
if (fileLength < (offset + buffMax)){
  buffMax = fileLength - offset;
  if(buffMax <= 0) {
    Serial.printf("attempt to seek past end of file.  File is %d bytes line, seeking to %d.\n",fileLength,offset);
  }
else { 
  f.read(buffer,buffMax);
  buffer[buffMax] = '\0'; // add null as end marker
  char * line = strtok(buffer,"\n");
  int lineNumber = 5;
  while(line){
     Serial.printf(" line:%04d = %s\n",lineNumber++,line);
     line = strtok(NULL,"\n"); //subdivide buffer at newline boundaries
  }
  ```

Chuck.

mehrdad987, 2019-08-24:
GREAT explanation
Unfortunately, the characters of each line will be different
I think I should change the way of sending my SDs file through the web

closed 2019-10-23 by mehrdad987