espressif/arduino-esp32 #3129: Esp32 rebooting continuously

wonnachai2, 2019-08-23:
I brought  ESP-32 about 4 piece from aliexpress last week. I have finshed upload  but I got a poblem that My Esp32 rebooting continuously.(I uploaded blank code.) Following the picture.

[image]

wonnachai2, 2019-08-23:
I brought ESP - 32  2019 Version on the Aliexpress.

stickbreaker, 2019-08-23:
@wonnachai2 What did you expect it to do?

Chuck.

wonnachai2, 2019-08-23:
@stickbreaker  First I uploaded blynk code and got a problem ESP-32 reboot continuously then I uploaded any code that got a same problem even blank code ESP-32 still reboot continuously Following the picture. I don’t know how can i fix it. Because The 4 pieces of ESP32 that I bought had a same poblem. 🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏🙏

wonnachai2, 2019-08-23:
@atickbreaker I guess that this is a 2019 Version of ESP-32 (Newest) but board esp-32 in Arduino IDE is older. I think right or wrong?

lbernstone, 2019-08-23:
See #3036

wonnachai2, 2019-08-23:
@lbernstone  Where is the Local of esp32/tools folder. I don't found it.

lbernstone, 2019-08-23:
Since you did not follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md), I do not know what operating system you are running.  Modern OSes have the ability to search for a file.

thylive, 2019-09-01:
have you solve the problem?

wonnachai2, 2019-09-01:
@thylive  No , I can't solve it.

atanisoft, 2019-09-01:
@wonnachai2 We can't help you solve this without you following the [issue template](https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md). You also need to be *VERY* explicit in which esp32 module you are using, there is no module named "esp32 2019 version". What exact one did you purchase on aliexpress? What do you have connected to the esp32 module?

There are more questions but until you answer the above and follow the issue template we can not help you resolve your issue.

@thylive If you have a similar issue I'd suggest not hijacking this one and create your own for tracking and please follow the issue template so we can help you with your issue.

Also note that a lot of us are available [here](https://gitter.im/espressif/arduino-esp32) to assist.

Kameeno, 2019-09-05:
Try by changing the flash mode from QIO to DIO

wonnachai2, 2019-09-06:
I try all lof method but it still same poblem , Now I buy new ESP 32 but I get a same poblem. (I notice that may old ESP32(can use) have a blue point on board) but my all of new esp 32(can not use) have a yellow point on board . @thylive Does your borad have a yellow point on board ?

thylive, 2019-09-09:
> I try all lof method but it still same poblem , Now I buy new ESP 32 but I get a same poblem. (I notice that may old ESP32(can use) have a blue point on board) but my all of new esp 32(can not use) have a yellow point on board . @thylive Does your borad have a yellow point on board ?

I do’t think it‘s yellow point‘s matter,if you use esptools to chick this chips , you will find that the cpu is 160mhz ,  but others  are 240 mhz.

check you chip , is there 2019 on it? i think this is  2019 version .

Kameeno, 2019-09-09:
I don’t know why but the aliexpress boards with esp32 they have lot of problems, 
First, I was able to run only with the DIO flash mode,
2nd they have lot of problem of soldering.. sometimes I need to resolver better the module pads

atanisoft, 2019-09-09:
> I don’t know why but the aliexpress boards with esp32 they have lot of problems,

You get what you pay for often... There are a number of sellers on AliExpress and not all are equal when it comes to quality control.

So far in this issue there has not yet been one definitive reference to the "bad board" just a bunch of vague references to a board that doesn't fit with any of the known boards. It is very likely this is a poorly manufactured clone of another board but without seeing exactly what board it will be very difficult to suggest a proper answer to the original issue being reported.

closed 2019-11-22 by stale[bot]

devitalii, 2023-12-21:
ESP32 mini S2 had this problem (still has on some boards from ebay/amazon).
The issue was restarting evey 1 second or so when using WiFi (broadcast a wifi network).

Solution from other forums: One capacitor on boards is not soldered right to GND.
Some newer boards are working fine. See below.
I tried to solder a blob of solder a 1/4 inch wire to EN pin on board (also tried 1 inch coiled wire), and it helped, esp32 is not resetting itself anymore and works fine.

Bad module was from amazon, good from Aliexpress (12.2023 order).

[image]