espressif/arduino-esp32 #3132: Error when using with MQTT PubSubClient package with ESP32.

earyzhe, 2019-08-24:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      disabled
Upload Speed:					115200
Computer OS:                                          Mac OSX

### Description:
When using with the PubSubClient package, errors are thrown.

The error message that is thrown is

>[D][WiFiClient.cpp:509] connected(): Disconnected: RES: -1, ERR: 104

When moving around the order in which the  "mqttClient.loop()" function fires in the main loop it changes the error.

```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <PubSubClient.h>
#include "DHT.h"
#include <ESP.h>

const char* SYSTEM_ID = "higrow-esp32-01";

char ssid[] = "(Don't mention the war)";
char password[] = "xxxxxxxxxxxxxxx";

#define LOOP_DELAY_SECS 600
#define PUMP_DELAY_MILLI 100

#define MQTT_SERVER "xxxxxxxxxx"
#define MQTT_PORT xxxxxxx
#define MQTT_USER "xxxxxxxf"
#define MQTT_PASSWORD "xxxxxxx"
#define MQTT_SERIAL_PUBLISH_TEST "test"
#define MQTT_SERIAL_PUBLISH_PLANTS "plants/berlin/oderstrasse/andrew"
#define MQTT_SERIAL_PUBLISH_CPU "things"
#define MQTT_SERIAL_PUBLISH_PLACE "places/berlin/oderstrasse/andrew"

extern uint8_t temprature_sens_read();

WiFiClient wifiClient;
PubSubClient mqttClient(wifiClient);

void setup() {
  
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  setupWifi();
  setupMqtt();
  delay(1000);
}

void loop() {
  mqttClient.loop(); 
  checkWifi();
  delay(2000);
}

void checkWifi() {
  if (WiFi.status() != WL_CONNECTED) {
    setupWifi();
    setupMqtt();
  }
}

void setupMqtt()
{

  Serial.print("MQTT Keep alive time is: ");
  Serial.print(MQTT_KEEPALIVE);
  Serial.println("s");
  // Loop until we're reconnected
 
    // Attempt to connect
    if (mqttClient.connect("coskc", MQTT_USER, MQTT_PASSWORD))
    {
      Serial.println("MQTT connected");

      //Once connected, publish an announcement...
      mqttClient.publish(MQTT_SERIAL_PUBLISH_TEST, SYSTEM_ID);
    }
    else
    {
      Serial.print("failed, rc=");
      Serial.print(mqttClient.state());
      Serial.println(" try again in 5 seconds");
    }
 while (!mqttClient.connected()) {

    Serial.print("Attempting MQTT connection with details: ID");
    Serial.print(SYSTEM_ID);
    Serial.print(" and  username: ");
    Serial.print(MQTT_USER);
    Serial.print(" and password ");
    Serial.println( MQTT_PASSWORD);

  }
  Serial.println("Finished setting up MQTT");
  delay(1000);
}

void setupWifi()
{
  // We start by connecting to a WiFi network
  WiFi.disconnect();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  for (int k = 0; k < 25; k++) {
    if (WiFi.status() != WL_CONNECTED)
    {
      delay(500);
      Serial.print(".");
    } else {
      break;
    }
  }

  if ( WiFi.status() != WL_CONNECTED )
  {
    Serial.println("Connecting to WiFi fail.");
    ESP.restart();
  } else {

    randomSeed(micros());
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    WiFi.printDiag(Serial);
    if (wifiClient.connect(MQTT_SERVER, MQTT_PORT))
    {
      Serial.print("Wifi client connected to ");
      Serial.println(MQTT_PORT);
    }
    Serial.println("Finished setting up wifi");
    delay(1000);
  }
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 

10:04:57.145 -> [W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
10:04:57.181 -> Connecting to (Don't mention the war)
10:04:57.253 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
10:04:57.253 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
10:04:57.467 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
10:04:57.501 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
10:04:57.501 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: xxxxxxx, MASK: 255.255.255.0, GW: 192.168.178.1
10:04:57.744 -> .
10:04:57.744 -> WiFi connected
10:04:57.744 -> IP address: 
10:04:57.744 -> xxxxxxxxxxx
10:04:57.778 -> Mode: STA
10:04:57.778 -> Channel: 11
10:04:57.778 -> SSID (23): (Don't mention the war)
10:04:57.778 -> Passphrase (20): xxxxxxxx
10:04:57.778 -> BSSID set: 0
10:04:57.953 -> Wifi client connected to 11968
10:04:57.953 -> Finished setting up wifi
10:04:58.968 -> MQTT Keep alive time is: 3600s
10:04:58.968 -> MQTT connected
10:04:58.968 -> Finished setting up MQTT
10:05:00.966 -> [D][WiFiClient.cpp:509] connected(): Disconnected: RES: -1, ERR: 104

```

atanisoft, 2019-08-24:
That isn't so much an error message but instead a debug log statement stating that the connection has been closed and is not usable. The PubSubClient should likely reconnect if it needs to automatically.

atanisoft, 2019-08-24:
The err: 104 translates to connection reset by peer. In other words the other end closed the connection.

earyzhe, 2019-08-25:
Hey, thanks for your help!

Well, I'm quite sure all my credentials are correct. And I have other devices connected to this broker.

The message given at the broker when I try to connect is :

> 2019-08-25 06:33:36: New connection from xx.xxx.xx.xx on port 11968.
> 2019-08-25` 06:33:37: Socket error on client <unknown>, disconnecting.

Just really confused about this. I guess this means some info not getting though currently. Have any Ideas?

atanisoft, 2019-08-25:
This is likely a bug in the PubSubClient code. I'd suggest moving this issue over to that repo for debugging since that library is not maintained in this repo.

earyzhe, 2019-08-25:
OK thanks :)

closed 2019-11-07 by stale[bot]