espressif/arduino-esp32 #3133: IllegalInstruction crash with Wire.begin()

manuelbl, 2019-08-24:
### Hardware:
Board:							WEMOS ESP32 OLED and also unbranded dev board with ESP32-D0WDQ6
Core Installation version:			1.0.1
IDE name:						Arduino IDE and also PlatformIO
Flash Frequency:					80MHz and also 40MHz
PSRAM enabled:                                      no
Upload Speed:					230,400
Computer OS:                                          macOS 10.14 and also Windows 10

### Description:
A very simple sketch (see below) crashes immediately during boot, most likely before reaching `setup()`.

If `Wire.begin(5, 6)` is changed to `Wire.begin(-1, -1)`, it no longer crashes and works as expected. (-1 are the default values.)

I've also tried to debug it with PlatformIO and an Esp-Prog adapter. The debugger nicely works with the working sketch. So the overall setup is correct. But for the crashing sketch, it fails to connect. OpenOCD either warns `Empty flash mapping!` (see OpenOCD output below) or it prints a long and most likely corrupt flash map. 

### Sketch: 

```cpp
#include <Arduino.h>
#include <Wire.h>

void setup() {
    Wire.begin(5, 6);
    Serial.begin(115200);
}

void loop() {
    Serial.println("Alive");
    delay(500);
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d177c: ffffffff ff2f0000 ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9910: ffffffff ffffffff ffffffff
GGuru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
```

### OpenOCD output when debugging:

```
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : accepting 'gdb' connection from pipe
Error: No symbols for FreeRTOS
Warn : Received debug request on stalled active core 0.
Info : Target halted. PRO_CPU: PC=0x40000400             APP_CPU: PC=0x00000000 (active)
Info : Target halted. PRO_CPU: PC=0x40000400             APP_CPU: PC=0x00000000 (active)
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Info : Target halted. PRO_CPU: PC=0x40000400             APP_CPU: PC=0x00000000 (active)
Error: Failed to get flash size!
Info : Target halted. PRO_CPU: PC=0x40000400             APP_CPU: PC=0x00000000 (active)
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

atanisoft, 2019-08-24:
Gpio 6 is connected to the flash chips and is not usable.

manuelbl, 2019-08-24:
Oops, yes indeed. The I2C wires are in fact connected to pin 5 and 16 and I misread the silk screen.

closed 2019-08-24 by manuelbl