espressif/arduino-esp32 #3134: ESP32 reboots on WiFi.softAP(ssid,password); or WiFi.enableAP(true); without error.

zekageri, 2019-08-26:
So i'am learning on my custom esp32 "board" recently. 
It is a **bare ESP32-WROOM-32 chip with 16mb flash**.
I have built a "board" for it from modules around it, to a breadboard.
I have several things on this board including

    -An lcd display 
    -Some I/O expander
    -A LAN 8720PHY ethernet board
    -A digital Potentiometer
    -A ds3231 RTC module
    -An EN125 ENTESLA RFID module
    -A key matrix
etc...

My power supply is an **EMG 18144 adjustable 0-40v and 0-5A DC power supply**.
Similar to this but mine is 5Amper:
[https://www.radiomuseum.org/r/elektronik_dc_power_supply_18135_vol.html](url)

**The issue:**
When calling either:

  - WiFi.softAP(ssid,password, WiFiChannel, 0, 1);
   - WiFi.softAP(ssid,password);
   - WiFi.mode(WIFI_AP);
   - WiFi.enableAP(true);

With or without delay after the functions,
the ESP RESTARTS _without a single error message_ on the serial monitor with
**build_flags = -DCORE_DEBUG_LEVEL=5 on the platformio.ini** file.

I have capacitors everywhere on the circuit.
One on the power supply pins, one on the Display POWER in pins and one on the ESP32's power pins.

The rest of the code works fine and the softAp was worked fine until today.
I'am using PlatformIO IDE on Vscode with latest libraries and latest arduino framework.

And these libraries:
 #include <Arduino.h>          // FRAMEWORK
#include <WebAuthentication.h>

/** WIFI LIBRARIES **/
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebSocketsServer.h>
#include <ESPmDNS.h>                // NOT WORKING PROPERLY!!!!
#include <WiFiUdp.h>
#include <ModbusIP_ESP8266.h>  // EXPERIMENTAL COMMUNICATION WITH PLC
/** I/O EXPANDER LIBRARY **/
#include "PCF8574.h"
#include <MCP23008.h>
/** KEY MATRIX LIBRARY **/
#include <tca8418.h>
/** ETHERNET COMMUNICATION LIBRARY **/
#include <Ethernet.h>
/** OVER THE AIR WEB UPDATE BOTH IP AND HTPP **/
#include <ArduinoOTA.h>
#include <Update.h>
/** DIGITAL POTENTIOMETER **/
#include <Systronix_AD5274.h>
/** SPIFFS LIBRARIES **/
#define FS_NO_GLOBALS
#include <FS.h>
#include "SPIFFS.h"

#include <Wire.h>                // I2C
#include <Ticker.h>              // Uptime
/** REAL TIME LIBRARI **/
#include <ESP32-hal-I2C.h>
#include <RtcDS3231.h>
//#include <EepromAt24C32.h>
/** SPI TFT LIBRARI AND DEFINES **/
#include <TFT_eSPI.h>
#include <SPI.h>
#include <JPEGDecoder.h>

/** OWN LIBRARIES **/
#include <sajat.h>                // ETHERNET HANDLES
#include <OwnLib.h>            // DECLARATIONS
#include "Pages.h"                // HTML PAGES
#include <Handles.h>           // ALL HANDLES
#include <I2C_Things.h>       // I2C FUNCTIONS
#include <DisplayUI.h>         // TFT FUNCTIONS
#include <RFID_Dolgok.h>    // RFID FUNCTIONS
#include <TCP_Requests.h>  // TCP/IP HANDLES
#include <SPIFFS_Dolgok.h> // SPIFFS FUNCTIONS

**What i tried:**
_-I tried palcing the WiFi.softAP(ssid,password);
To everywhere on my sketch.
To the setup, to the loop with a turn on flag.
-I tried to disconnect the wifi before the softap.
-I tried (150) delay after the softap.
-I tried softap config ( static ip ) before and after the softap.
-I tried with different power supply.
-I tried without capacitors on the circuit._

**-I tried it with a following bare async sketch(Result is the same, ESP REBOOTS without ERROR):**

```
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "ProbaAP";
const char* password = "12345678";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

void setup(){
Serial.begin(115200);
WiFi.softAP(ssid,password);

server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    server.onNotFound(notFound);
    server.begin();
}

void loop(){

}
```

**-With simple Sync Webserver (Result is the same, ESP REBOOTS without ERROR):**

```
#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>

/* Put your SSID & Password */
const char* ssid = "ESP32";  // Enter SSID here
const char* password = "12345678";  //Enter Password here

WebServer server(80);

void handle_OnConnect() {
  server.send(200, "text/html", "Hello server!"); 
}
void handle_NotFound(){
  server.send(404, "text/plain", "Not found");
}

void setup() {
  Serial.begin(115200);
  WiFi.softAP(ssid, password);
  delay(100);
  server.on("/", handle_OnConnect);
  server.onNotFound(handle_NotFound);
  
  server.begin();
  Serial.println("HTTP server started");
}
void loop() {
  server.handleClient();
}
```

[image]
[image]

CircuitSetup, 2019-09-27:
You should run the serial output of the crash through the ESP32 Exception Decoder. 

Regardless, this is what has worked for me:
```
  WiFi.enableAP(true);
  delay(100);
  WiFi.softAPConfig(apIP, apIP, netMsk);
  WiFi.softAP(softAP_ssid, softAP_password, channel);
  delay(500); // Without delay the IP address is sometimes blank
```

zekageri, 2019-09-30:
> You should run the serial output of the crash through the ESP32 Exception Decoder.
>
> Regardless, this is what has worked for me:
>
> ```
>   WiFi.enableAP(true);
>   delay(100);
>   WiFi.softAPConfig(apIP, [...]

Thank you for your suggestions, but as i mentioned the ESP reboots without a single error message on the serial. ^^ 
The problem was at my circuit. 
The pull up resistors on the i2c bus are touched each other.

Sorry for not closing this before.

closed 2019-09-30 by zekageri