espressif/arduino-esp32 #3136: .concat, different behaviour in 1.0.3rc2

joseitor, 2019-08-26:
I have just update the ESP32 firm to last release 1.0.3rc2 and the behaviour of concatenation function has changed, and I cannot find any advice in the release notes about it.

Using this code:
`    String responseParams = "";
    char buffer1[sizeof(long) * 8 + 1];
    char date[6] = "";
    strcat(date, ltoa(260819, buffer1, 10));
    char Time[6] = "";
    strcat(Time, ltoa(140530, buffer1, 10));
    responseParams.concat(date + HASH + Time);
    Serial.println(date);
    Serial.println(Time);
    Serial.println(responseParams.c_str());`

In 1.0.2 it prints "260819#140530"
but in 1.0.3rc2 ir prints "260819140530#140530"

Obviously there is some bug there, or in other case the change in the function behave must be notice.

joseitor, 2019-08-26:
But in the version 1.0.2 works perfectly...it mustn't have change anything...

me-no-dev, 2019-08-26:
Oh, you forgot to zero-terminate the strings and it happened that `date` and `time` are right next to each other in the memory and there is a zero after `time`. So it printed both date and time, found a zero and stopped, then printed the hash and then the time again (till the zero).
Make your char arrays of len 7 instead and add `date[6]=0;` to make sure that you have the strings terminated.

closed 2019-08-26 by me-no-dev