espressif/arduino-esp32 #3155: WiFiClientSecure SSL certificate question

Adrianotiger, 2019-08-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
Hi, I am trying to connect to my server with a certificate. I am able to connect only if I use my root or parent certificate, not the one released for my server:
[image]

I am asking me if it is just a bug or if it is normal. I didn't used the other 2 certificates because it is easy to get my certificate with just 3-4 lines of PHP-code.

**As said before, maybe this is not a bug.** But it is normal that it doesn't work with the last certificate. With the parent certificate (Let's Encrypt) and the root certificate (DST Root CA X3) I am able to connect to my server. I can use the parent certificate with just 1 more php-code but for the root certificate, I need more...

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
This is the php code to get a server certificate (so the ESP can download it if the SSL is old)
```php
$g = stream_context_create (array("ssl" => array("capture_peer_cert" => true)));
$r = stream_socket_client("ssl://MYWEBPAGE:443", $errno, $errstr, 30, STREAM_CLIENT_CONNECT, $g);
$cont = stream_context_get_params($r);
$certinfo = openssl_x509_parse($cont["options"]["ssl"]["peer_certificate"]);
openssl_x509_export($cont["options"]["ssl"]["peer_certificate"], $cert);
echo $cert;
```

### Debug Messages:
```
[V][HTTPClient.cpp:235] beginInternal(): url: https://MYWEBPAGE/api/1.0/
[D][HTTPClient.cpp:276] beginInternal(): host: MYWEBPAGE port: 443 url: /api/1.0/
[V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 240116
[V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[V][ssl_client.cpp:90] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:99] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:112] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:177] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:192] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:33] handle_error(): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -9984
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
[V][ssl_client.cpp:245] stop_ssl_socket(): Cleaning SSL connection.
[D][HTTPClient.cpp:1018] connect(): failed connect to MYWEBPAGE:443
[W][HTTPClient.cpp:1318] returnError(): error(-1): connection refused
```

closed 2019-11-11 by stale[bot]

savejeff, 2023-03-19:
this problem still persists. the lowest certificate does not work 
Testen on ESP32-S3.

its not a huge problem, but its something to easily stumble over. maybe it would be good to add the information to the example ino

JpEncausse, 2024-04-24:
Thank you SO MUCH for this explanation I was stuck for months on that !
Now with parent Let's Encrypt certificate it works !