espressif/arduino-esp32 #3157: [W][AsyncTCP.cpp:717] _poll(): pcb is NULL

mcuadros, 2019-08-29:
### Hardware:
Board:							lolin_d32, esp32_minikit
Core Installation version:			1.0.2 
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:				115200
Computer OS:                                        Arch

### Description:
I have a random issue related to TCP connections, in this case trying to connect to an MQTT server with AsyncMQTTClient after receiving the event `STA_GOT_IP`, the connection fails, and ASyncTCP.cpp throws a warning `[W][AsyncTCP.cpp:717] _poll(): pcb is NULL`

If during this connection period I do a ping, I don't get any answer, but after a few seconds doing the ping the warning disappears and the connection is established. If a don't do any ping, the connection never success. 
Also, in some random boots, this works perfectly.

The code I am using is a example from the MQTT client:
https://github.com/marvinroger/async-mqtt-client/blob/master/examples/FullyFeatured-ESP32/FullyFeatured-ESP32.ino

I tried with different boards.

This issue is similar to: https://github.com/espressif/esp-idf/issues/2184

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*
This example uses FreeRTOS softwaretimers as there is no built-in Ticker library
*/

#include <WiFi.h>
extern "C" {
	#include "freertos/FreeRTOS.h"
	#include "freertos/timers.h"
}
#include <AsyncMqttClient.h>

#define WIFI_SSID "yourSSID"
#define WIFI_PASSWORD "yourpass"

#define MQTT_HOST IPAddress(192, 168, 1, 10)
#define MQTT_PORT 1883

AsyncMqttClient mqttClient;
TimerHandle_t mqttReconnectTimer;
TimerHandle_t wifiReconnectTimer;

void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
}

void WiFiEvent(WiFiEvent_t event) {
    Serial.printf("[WiFi-event] event: %d\n", event);
    switch(event) {
    case SYSTEM_EVENT_STA_GOT_IP:
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
        connectToMqtt();
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        Serial.println("WiFi lost connection");
        xTimerStop(mqttReconnectTimer, 0); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
		xTimerStart(wifiReconnectTimer, 0);
        break;
    }
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 2);
  Serial.print("Subscribing at QoS 2, packetId: ");
  Serial.println(packetIdSub);
  mqttClient.publish("test/lol", 0, true, "test 1");
  Serial.println("Publishing at QoS 0");
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
  uint16_t packetIdPub2 = mqttClient.publish("test/lol", 2, true, "test 3");
  Serial.print("Publishing at QoS 2, packetId: ");
  Serial.println(packetIdPub2);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");

  if (WiFi.isConnected()) {
    xTimerStart(mqttReconnectTimer, 0);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
  Serial.print("  qos: ");
  Serial.println(qos);
}

void onMqttUnsubscribe(uint16_t packetId) {
  Serial.println("Unsubscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("Publish received.");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  mqttReconnectTimer = xTimerCreate("mqttTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToMqtt));
  wifiReconnectTimer = xTimerCreate("wifiTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToWifi));

  WiFi.onEvent(WiFiEvent);

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);

  connectToWifi();
}

void loop() {
}
```

### Debug Messages:
```
Connecting to Wi-Fi...
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2
[WiFi-event] event: 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 10.3.167.132, MASK: 255.255.255.0, GW: 10.3.167.199

10.3.167.132
Connecting to MQTT...
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
[W][AsyncTCP.cpp:717] _poll(): pcb is NULL
```

closed 2019-11-11 by stale[bot]

dr3adx, 2023-02-26:
WHY IS THIS CLOSED

me-no-dev, 2023-02-26:
it's external library @dr3adx

dr3adx, 2023-02-27:
> it's external library @dr3adx

I'm using AsyncHTTPReq lib that uses asyncclient and i keep getting pcb == NULL error when sending A COUPLE simultaneous requests, why is this? Its not even that many reqs, maybe 6 per second.

bertmelis, 2023-02-27:
pcb == NULL means that there is not yet a TCP connection.
It's just a **warning** from the onPoll handler, which is called regularly from the moment you **try** to connect.

dr3adx, 2023-02-27:
> pcb == NULL means that there is not yet a TCP connection. It's just a **warning** from the onPoll handler, which is called regularly from the moment you **try** to connect.

eventually i will get: assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

any idea why this happens?

bertmelis, 2023-02-27:
no

dr3adx, 2023-02-27:
> it's external library @dr3adx

@me-no-dev please check https://github.com/me-no-dev/AsyncTCP/issues/161, this crash is so frustrating it killed my dreams of async reqs

Gornerrrrr, 2023-07-06:
Same problem, and sometimes its run.

rljonesau, 2023-12-06:
After much frustration, I have not found a solution, but have found what causes this to manifest.

It is easily reproducible if you lose the AP connection, eg forcibly switched off, then switch to another AP.
The other AP is connected OK, but the connect fails in this manner for a while then the connection is given up, and a later retry still gives PCB=NULL
After several minutes, it eventually comes good.

It seems to be LWIP "in a mood", especially after the beacon timeout situation and a fresh different AP connection.

One almost thinks the only way to get a timely response is reboot the ESP, but how uncouth is that!

me-no-dev, 2023-12-06:
You need to make sure that the APs you connect to are on the same channel, else ESP-NOW will fail when switching