espressif/arduino-esp32 #3160: Webserver Hangs on Chrome after "New Client" but works smooth in Firefox

kapyaar, 2019-08-30:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3rc2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600?
Computer OS:                                        Windows 10

### Description:
I am working on a simple webserver. When requested via Chrome, there is a consistent 4-5 second delay after log prints
`[V][WebServer.cpp:286] handleClient(): New client`

If requested via Firefox (and may other browsers I tested), This delay is not there, and all goes smooth. There are similar posts talking about this issue, yet does not seem to have been resolved. 

1. https://github.com/espressif/arduino-esp32/issues/1931
2. https://github.com/espressif/arduino-esp32/issues/1921
3. https://github.com/espressif/arduino-esp32/issues/1023

I have something else to add here. Hence starting this new issue. I see differences in the header requests made by Chrome and FF, EE,  Safari [Mac and iOS], Chrome[iOS], . Notably

1. headerName: Accept
2. headerName: Accept-Language

See Full headers below. May be this can give some insight??

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))

HelloServer Example

### Debug Messages:

When using Chrome
```
[V][WebServer.cpp:286] handleClient(): New client

================5 Second Delay here=======================

[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.0.14
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: max-age=0
[V][Parsing.cpp:219] _parseRequest(): headerName: Upgrade-Insecure-Requests
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/76.0.3809.132 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,pl;q=0.8
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
```

When using FF -> Works Smoothly. Page loads right away.
```
[V][WebServer.cpp:286] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.0.14
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:68.0) Gecko/20100101 Firefox/68.0
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[... 69 lines cut ...]

When using Chrome on Iphone, Works Smoothly. Page loads right away.
```
[V][WebServer.cpp:286] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.0.14
[V][Parsing.cpp:219] _parseRequest(): headerName: Upgrade-Insecure-Requests
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (iPhone; CPU iPhone OS 12_2 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) CriOS/76.0.3809.123 Mobile/15E148 Safari/605.1
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-us
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128

```

ElToberino, 2020-04-20:
This still doesn't seem to be solved. I'm facing the same issue.

VojtechBartoska, 2022-05-04:
I'm closing this issue as expired. If it's needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska