espressif/arduino-esp32 #3162: I2C read/write operation blocks a task

rammak, 2019-09-01:
### Hardware:
Board:							ESP32-devkit
Core Installation/update date:		1.0.2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I have attached a MPU-6050 accelerometer breakout module with ESP32-devkit. I'm creating a new task to handle MPU related functions while my loop task handles other functions. The MPU task is at priority 2 while the default loop task is at priority 1. In my project, the accelerometer stops responding sometimes and to remedy that I'm reinitializing the sensor by power cycling it.

The problem is that if I initialize the accelerometer successfully and it stops working between successful initialization and reading a value from it, the task gets stuck at that point and it doesn't let the default loop task to run so the default task is blocked indefinitely. Ideally, i2c timeout is set to 50ms so after 50ms the MPU task should give up reading and yield but it does not happen.

There are 2.2k pullups connected on SDA and SCL line in the accelerometer breakout board. I have tried manually disconnecting the accelerometer's power to simulate a fault condition and it doesn't cause a problem if I read after failed initialization, the read function will simply not get executed. But if the initialization and checking are successful and power is disconnected between that point and while reading is happening or before reading, it gets stuck. I can see the state of the task in that stuck phase which is 0(running) for MPU task and 1(ready to run) for default loop task.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup() {
  serial_init();

  //  initialize peripherals

  //initialize BLE

  // create a new RTOS task to handle MPU
  xTaskCreatePinnedToCore
              (mpu_task,         /* Task function. */
              "mpu_task",       /* String with name of task. */
              10000,            /* Stack size in bytes. */
              NULL,             /* Parameter passed as input of the task */
              2,                /* Priority of the task. Loop task's priority is 1 so this should be higher*/ 
              &mpu_task_handle, /* Task handle. */
              1);            
}

void loop() {
  BLE_handle();
  taskYIELD();  // may not be necessary
}

void mpu_task(void * parameter) {
  aSerial.v().println("MPU: Task created ");
  MPU_init();

  while(1) {
    if(!MPU_handle()) {
      MPU_init();
    }
    taskYIELD();
  }
}

bool check_MPU()
{
    uint8_t value;

    Wire.beginTransmission(MPU6050_ADDRESS);
    Wire.write(MPU6050_REG_WHO_AM_I);
    Wire.endTransmission();

    Wire.beginTransmission(MPU6050_ADDRESS);
    Wire.requestFrom(MPU6050_ADDRESS, 1);
    value = Wire.read();
    Wire.endTransmission();

    if(value != MPU6050_ADDRESS)
    {
      MPU_online = false;
      return false;
    }
    else
    {
      MPU_online = true;
      return true;
    }
}

void MPU_init() {

  pinMode(MPU_POWER_PIN, OUTPUT);
  digitalWrite(MPU_POWER_PIN, LOW);
  vTaskDelay(1000);
  digitalWrite(MPU_POWER_PIN, HIGH);
  vTaskDelay(1000);
  aSerial.vv().print("MPU: Timeout : ").println(Wire.getTimeOut());
  int i = 2; // retry counts
  while(!mpu.begin(MPU6050_SCALE_250DPS, MPU6050_RANGE_16G) && i > 0)
  {
    aSerial.vv().println("MPU: Could not find a valid MPU6050 sensor, check wiring!");
    vTaskDelay(10);
    i--;
  }

  if(i == 0) {
    aSerial.v().println("MPU: Initialization failed");
    MPU_online = false;
  }
  else
  {
    MPU_online = true;
    aSerial.v().println("MPU: Initialization successful");
    mpu.setDLPFMode(MPU6050_DLPF_6);
  }
}

bool MPU_handle() {
  aSerial.vvv().print("MPU: Checking wiring");
  if(!check_MPU()) {    //  in case of MPU failure, return false to keep the cycle running
    aSerial.vv().println("MPU: MPU disconnected");
    return false;
  }
  aSerial.vvv().print("MPU: Check finished");
  // task gets stuck here is power gets disconnected
  temp_value = mpu.readTemperature();
  Vector rawAccel = mpu.readRawAccel();
  // doing some processing
  return true;
}
```

### Debug Messages:
```
```

stickbreaker, 2019-09-01:
@rammak Set CORE_DEBUG_LEVEL to "WARNING" and rerun, capture and post the serial debug log.  Lets see if the I2C subsystem detects any errors. The timeout will happen. The i2c subsystem will not lock in a loop.

Chuck.

rammak, 2019-09-02:
> 22:51:57.167 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
22:51:57.167 -> [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
22:51:57.235 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
22:51:57.235 -> [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
22:51:57.709 -> MPU: stack - 8204
22:51:57.709 -> MPU task State: 2
22:51:57.709 -> Arduino task State: 0
22:51:58.685 -> MPU: stack - 8204
22:51:58.718 -> MPU task State: 2
22:51:58.718 -> Arduino task State: 0
22:51:59.220 -> MPU: Timeout : 50
22:51:59.288 -> MPU: Could not find a valid MPU6050 sensor, check wiring!
22:51:59.288 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=0
22:51:59.288 -> [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
22:51:59.288 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=1, scl(22)=0
22:51:59.321 -> [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
22:51:59.690 -> MPU: stack - 8204
22:51:59.690 -> MPU task State: 2
22:51:59.690 -> Arduino task State: 0
22:52:00.698 -> MPU: stack - 8204
22:52:00.698 -> MPU task State: 2
22:52:00.698 -> Arduino task State: 0
22:52:01.300 -> MPU: Timeout : 50
22:52:01.300 -> MPU: Initialization successful
22:52:01.703 -> MPU: stack - 8204
22:52:01.703 -> MPU task State: 2
22:52:01.703 -> Arduino task State: 0
22:52:02.707 -> MPU: stack - 8204
22:52:02.707 -> MPU task State: 2
22:52:02.707 -> Arduino task State: 0
22:52:03.713 -> MPU: stack - 8204
22:52:03.713 -> MPU task State: 2
22:52:03.713 -> Arduino task State: 0
22:52:04.682 -> MPU: stack - 8204
22:52:04.717 -> MPU task State: 2
22:52:04.717 -> Arduino task State: 0
22:52:05.691 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=1, scl(22)=0
22:52:05.691 -> [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
22:52:05.724 -> 1
22:52:06.692 -> MPU: stack - 8204
22:52:06.692 -> MPU task State: 0
22:52:06.692 -> Arduino task State: 1
22:52:07.692 -> MPU: stack - 8204
22:52:07.692 -> MPU task State: 0
22:52:07.692 -> Arduino task State: 1
22:52:08.696 -> MPU: stack - 8204
22:52:08.696 -> MPU task State: 0
22:52:08.696 -> Arduino task State: 1
>

Thank you @stickbreaker. Here is the log after setting CORE_DEBUG_LEVEL to "WARNING". As it seems, it has something to do with the states of SCL and SDA pins. One thing I noticed is that the accelerometer board has an on-board power LED and I have attached the VCC of the accelerometer directly to a GPIO of MCU(the power consumption is quite low so it shouldn't be an issue) and I'm toggling the GPIO to power cycle the accelerometer. But sometimes when I disconnect the VCC from GPIO manually, the LED stays dimly on which means the accelerometer is getting some power from SDA and SCL lines. 

I have noticed that this doesn't happen every time I abruptly disconnect the power from the accelerometer but whenever it happens, the task gets stuck and sometimes I don't even see that error message on debug (other messages that get printed each second are from a Ticker ISR). I think this issue is related to that parasitic power.

stickbreaker, 2019-09-02:
An issue you could be running into, the esp32 CANNOT handle SCL stretching longer than 13.1ms when Cpu clock is at or above 80mhz. max stretch is 2**20 APB clocks. APB clock is equal to CPU clock or 80Mhz whichever is lower. 

If the Stretch exceeds this hard limit, a timeout error is generated and the transaction is aborted.

Your mpu might be drawing current from the i2c bus.

rammak, 2019-09-03:
> An issue you could be running into, the esp32 CANNOT handle SCL stretching longer than 13.1ms when Cpu clock is at or above 80mhz. max stretch is 2**20 APB clocks. APB clock is equal to CPU clock or 80Mhz whichever is lower.
> 
> If the Stretch exceeds this hard limit, a timeout error is generated and the transaction is aborted.
> 
> Your mpu might be drawing current from the i2c bus.
> 
> Try going through a more comprehensive reset cycle:
> 
> ```c++
> // power off the mpu with your cpu pin
> digitalWrite(powerPin,LOW);
> // shutdown Wire() and manual lock the bus low
> ~Wire(); // free, shutdown i2c.
> pinMode(SDA,OUTPUT);
> pinMode(SCL,OUTPUT);
> digitalWrite(SDA, LOW);
> digitalWrite(SCL,LOW);
> delay(100); // give mpu time to actually turn off.
> digitalWrite(powerPin,HIGH); // restart MPU
> pinMode(SCL,INPUT);
> pinMode(SDA,INPUT);
> Wire = TwoWire(0); // reinit Wire() object
> if(!Wire.begin(SDA,SCL,100000){ // use default pins and 100khz
>    Serial.printlin("Wire did not init, problem with bus");
> }
> ```
> 
> Chuck.

I have tried changing the init code but it still gets stuck. As I suspected before, the initializing code is fine since it doesn't get stuck there but mostly get stuck during reading values in MPU_handle function. I removed the two reading functions from there and the issue seems to go away and only reappear if I quickly connect and cut the supply which points to something being wrong during reading. 

I have also measured the voltage on SDA and SCL when it gets stuck and they mostly stay at around 2V mark. Right now the module is soldered to the board so I can't disconnect those lines but as you said the module is definitely getting some power from the data and clock lines.

rammak, 2019-09-10:
Ok. Finally, I solved the issue. Actually the problem was in the [library](https://github.com/jarzebski/Arduino-MPU6050/blob/master/MPU6050.cpp) that I used. It used while loop with Wire.available() which used to get stuck. I removed those waiting while loops and now it is working fine with low I2C frequency values(<100000).

closed 2019-09-10 by rammak