espressif/arduino-esp32 #3166: I_DELAY ULP macros not working

perseus086, 2019-09-01:
I want to make a led blink via ULP macros,
I use the following code:

```
#include "esp32/ulp.h"
#include "driver/rtc_io.h"

void ULP_BLINK_RUN(uint32_t us);

void setup() {
  ULP_BLINK_RUN(1000);
}

void loop() {
  esp_deep_sleep_start();
}
// ------------------------------------------------------------------------
void ULP_BLINK_RUN(uint32_t us) {
  ulp_set_wakeup_period(0, us);
  const ulp_insn_t  ulp_code[] = {
    I_WR_REG(RTC_GPIO_OUT_REG, 24, 24, 1),    // GPIO 4 --> HIGH
    I_DELAY(150000),
    I_WR_REG(RTC_GPIO_OUT_REG, 24, 24, 0),    // GPIO 4 --> LOW
    I_DELAY(150000),
    I_HALT()
  };

  // OUTPUTS
  const gpio_num_t gpio_outputs[] = {
    GPIO_NUM_2,
    GPIO_NUM_0,
    GPIO_NUM_4,
    GPIO_NUM_15
  };
  for (size_t i = 0; i < sizeof(gpio_outputs) / sizeof(gpio_outputs[0]); ++i) {
    rtc_gpio_init(gpio_outputs[i]);
    rtc_gpio_set_direction(gpio_outputs[i], RTC_GPIO_MODE_OUTPUT_ONLY);
    rtc_gpio_set_level(gpio_outputs[i], 0);
  }
  
  size_t size = sizeof(ulp_code) / sizeof(ulp_insn_t);
  ulp_process_macros_and_load(0, ulp_code, &size);
  ulp_run(0);
}
```

LED only turns on. I made other tests and seems that I_DELAY is not working.

What can be the problem here?

Jeroen88, 2019-09-02:
It did some experiments some time ago, so I am certainly no expert... However, the problem might be that if ULP_BLINK_RUN(1000) starts the ULP code every 1000 us, so 1000 times a second. Before it reaches the code to set the pin low, it already runs again and keeps the pin high.

The problem might also be that the ULP has 16 bit registers, and 150000 won't fit in 16 bits. I believe also that I_DELAY() is in ULP clock cycles. If 150000 would fit, the delay is 150000 / 8000000 = 0,01875 seconds, which might be to short to notice.

You can find more info [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/ulp.html)

perseus086, 2019-09-02:
ULP runs at 8MHz or at 150KHz?

I thought that we can run the program for any time until it finds a HALT instruction and then it restarts again the counter that will wake up the program again.
That means that even if it takes minutes to run it should not be started again by ulp_set_wakeup_period and interrupt the execution

> The program runs until it encounters a halt instruction or an illegal instruction. Once the program halts, ULP coprocessor powers down, and the timer is started again.

Anyway, you are totally right with the **max value of I_DELAY** it should be 65535 cycles and I think that is the cause that it is not doing the delay.

Jeroen88, 2019-09-02:
I think it runs at [8MHz](http://esp-idf.readthedocs.io/en/latest/api-guides/ulp_instruction_set.html#note-about-instruction-execution-time)

Yes, I think you are right, the counter starts again after a HALT

closed 2019-11-15 by stale[bot]