espressif/arduino-esp32 #3167: ESP32 core 0 panicked while generating interrupt

vbshightime, 2019-09-02:
### Hardware:
Board:							ESP 32 DEVKIT v1
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:				        115200
Computer OS:                                         Windows 10

### Description:
I am generating an interrupt on pin change to put ESp32 on configuration mode. In config mode i have couple of html pages which are required to configure the sensor nodes. otherwise the esp32 will get the values from sensor nodes which is running on core 0. But while I press the push button to gennerate an interrupt ESP is throwing following exception 

### Sketch:  
```cpp

void IRAM_ATTR ISR_int(){
     APTimer = millis();
     vTaskSuspend(xHandle);
     detachInterrupt(digitalPinToInterrupt(intPin));
     gatewayConfig=1;
  }

void setup() {
  Serial.begin(115200);
  Serial.print("Started");
  pinMode(intPin,INPUT);
  pinMode(ledpin, OUTPUT);
  digitalWrite(ledpin,LOW);
  //attachInterrupt(digitalPinToInterrupt(intPin), ConfigInt, FALLING);
  EEPROM.begin(512);
  attachInterrupt(digitalPinToInterrupt(intPin),ISR_int,CHANGE);
  xTaskCreatePinnedToCore(
    TaskSend
    ,  "TaskSendNRF"   
    ,  4096 
    ,  NULL
    ,  2  
    ,  &xHandle 
    ,  0);
}
void loop() {

    if(gatewayConfig){
          Serial.println("interruptGenerated");
          handleWebform();
      }

void TaskSend(void *pvParameters)  // This is a task.
{
  (void) pvParameters;
   
  //reconnectWiFi();
  for (;;) // A Task shall never return or exit.
  {     
   if ( radio.available()) {
     if(WiFi.status()==WL_CONNECTED){
    if(buff!=""){
        strcpy(buff,"");
      }    
       snprintf(messagePayload,WIFI_MESSAGE,MESSAGE_BODY,devCat.c_str(),macId.c_str(),sensor,battery);
        Serial.println(messagePayload);
        if(sizeof(messagePayload)>0){
                Serial.println("Entered HTTP");
               http":"NotConnected http");//zx
                http.addHeader("Content-Type" , "application/json");
                for(int attempt=0;attempt<5;attempt++)
                   {
                    //Serial.println("inside loop");    
                    int httpCode = http.POST(messagePayload);             //Send the request
                    String payload = http.getString();                  //Get the response payload
                    Serial.println(httpCode);   //Print HTTP return code
                    Serial.println(payload);
                    if(httpCode == HTTP_CODE_OK) {
                    break;        
                   }
                  vTaskDelay(100);  // one tick delay (15ms) in between reads for stability
                  }
                  http.end();
               }
        }
        else{
                 reconnectWiFi();
              }
    vTaskDelay(5);
  }
    vTaskDelay(1000);
  }
}

```

### Debug Messages:
```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401487a0: 02f01d00 23084136 20c01edf
Core 0 register dump:
PC      : 0x401487a7  PS      : 0x00050031  A0      : 0x400817e8  A1      : 0x3ffbdda0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x800e6682  A5      : 0x4008b774  
A6      : 0x00000000  A7      : 0x400862c4  A8      : 0x00000000  A9      : 0x3ffbbb20  
A10     : 0x00000000  A11     : 0x3ffbbea0  A12     : 0x800e6677  A13     : 0x3ffbc160  
A14     : 0x00000000  A15     : 0x3ffbc18f  SAR     : 0x00000011  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401487a7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x401487a7:0x3ffbdda0 0x400817e5:0x3ffbe1c0 0x40167777:0x3ffbc1b0 0x400e49aa:0x3ffbc1d0 0x4008a8fd:0x3ffbc1f0 0x400887f9:0x3ffbc210
```

### Decoded Messages:
```
PC: 0x401487a7: rtc_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 1965
EXCVADDR: 0x00000000

Decoding stack results
0x401487a7: rtc_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/rtc_module.c line 1965
0x40167777: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487
0x400e49aa: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 63
0x4008a8fd: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3412
0x400887f9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

CelliesProjects, 2019-09-02:
ROM functions are not allowed in a interrupt callback
> vTaskSuspend(xHandle);
     detachInterrupt(digitalPinToInterrupt(intPin));

Just set a flag and handle the rest in a task.

vbshightime, 2019-09-03:
I had put vTaskSuspend(xHandle) outside the ISR. But I was still getting the same issue and the problem doesnt lie here. The core was throwing exception just after it is trying to connect to AP. as the other task   might be connecting to STA at the same time.

> void TaskSend(void *pvParameters)  // This is a task.
> {
>   (void) pvParameters;
>    
>   //reconnectWiFi();
>   for (;;) // A Task shall never return or exit.
>   {     
>    if ( radio.available()) {
>      if(WiFi.status()==WL_CONNECTED){

> WiFi.mode(WIFI_AP);
>       delay(100);
>       Serial.println(WiFi.softAP(ssidAPConfig,passAPConfig) ? "Configuring softAP" : "kya yaar not connected");
>       WiFi.softAPConfig(apIP, apIP, netMsk);    

 I solved this problem by disconnecting to WiFi once and then going with the AP process

closed 2019-09-05 by vbshightime