espressif/arduino-esp32 #3169: WDT reset

vignesh13404387, 2019-09-03:
Hello,

I faces some issues when try to blink led without using delay() and used "if ((millis() - lastReport) >= 1000)" like this on the core 0 with the use of xTaskCreatePinnedToCore() function. It throws the errors

E (10169) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10169) task_wdt: - IDLE0 (CPU 0)
E (10169) task_wdt: Tasks currently running:
E (10169) task_wdt: CPU 0: get reading
E (10169) task_wdt: CPU 1: IDLE1
E (10169) task_wdt: Aborting.
abort() was called at PC 0x400d5bd7 on core 0

I am also running core 1 to print "hello world" in the serial monitor.

Please, anyone, help to understand it.

Jeroen88, 2019-09-03:
What if you try this

```
void print_time(void * parameter) {
  uint32_t lastReport = millis();
  for (;;) {
    if ((millis() - lastReport) >= 1000) {
      Serial.println(millis());
      lastReport = millis();
    }
    delay(1);
  }
}
```

lastReport should be initialized to millis() not to zero

The delay(1) is to prevent the wdt to kick in. I am not very familiar with tasks, but there should be a better way with tasks that run on a timer instead of the active waiting in the for() loop

vignesh13404387, 2019-09-03:
It worked and thank you for your valuable comment @Jeroen88 .  Would you tell why that delay(1) is used in detail?  I am not aware of that.

lbernstone, 2019-09-03:
If you are going to create tasks, you must manage their priorities properly.  Having all your tasks at the same priority, and then never releasing focus (with delay), means that the processor cannot context switch.

vignesh13404387, 2019-09-03:
Hello, @lbernstone  here I have two tasks both running in separate cores.  Is it cause problem if I have the same priority?

atanisoft, 2019-09-03:
@vignesh13404387 As @lbernstone mentions, you must have a call to vTaskDelay() or delay() within your tasks to allow the task scheduler to run other tasks at different priorities (even same if the case may be, and the IDLE[0,1] tasks).

atanisoft, 2019-09-03:
Also the Ticker library would be a better approach for your timed interval needs.

lbernstone, 2019-09-03:
Let me make this simple. Unless you are a skilled programmer, you MUST have a delay in your task loops.

closed 2019-09-03 by vignesh13404387