espressif/arduino-esp32 #3171: PlatformIO Installation Insrtuctions

jhoughjr, 2019-09-03:
Why are there no instructions to install via PlatformIO?
Why do "What is PlatformIO?
PlatformIO IDE
PlatformIO Core (command line tool)
Advanced usage - custom settings, uploading to SPIFFS, Over-the-Air (OTA), staging version
Integration with Cloud and Standalone IDEs - Cloud9, Codeanywhere, Eclipse Che (Codenvy), Atom, CLion, Eclipse, Emacs, NetBeans, Qt Creator, Sublime Text, VIM, Visual Studio, and VSCode
Project Examples" appear to having nothing framework specific?

beegee-tokyo, 2019-09-03:
This repository is focusing on ArduinoIDE. Instructions for PlatformIO can be found [here](https://platformio.org/platformio-ide)

lbernstone, 2019-09-03:
I intend to do some rework on that page once the latest dev stuff makes it into platform.io stable.  At that point platform.io will be the recommended method for using arduino as an esp-idf component.

atanisoft, 2019-09-03:
@beegee-tokyo This repo is *NOT* just for Arduino IDE. It is also leveraged by PlatformIO when the platform is espressif32 and framework is set to arduino (and by extension espidf, arduino as a combination framework supported by PIO dev versions).

@jhoughjr PlatformIO supports multiple target platforms, espressif32 (esp32) is only one of them. If you look at the [PlatformIO docs](https://github.com/espressif/arduino-esp32/blob/master/docs/platformio.md) these provide links to the general setup of PlatformIO including examples of use (the last link) which are specific to the espressif32 (esp32) platform.

This issue can likely be closed as it should have likely been asked over on [gitter](https://gitter.im/espressif/arduino-esp32) or on www.esp32.net instead. It does not describe an actual issue with the arduino-esp32 codebase nor does it follow the [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

closed 2019-11-16 by stale[bot]