espressif/arduino-esp32 #3172: Parse v 3.2 to Arduino ide

athleticmistud, 2019-09-03:
I am trying to connect to a Bluetooth speaker and send a 39 second audio file to the speaker.

I need it parsed for Arduino ide can someone help please to parse this v:3.2

Thanks

athleticmistud, 2019-09-05:
here is the data from serial monitor on the d32 here is what we did

i load code on d32 

then load our code on d1   only connection between the 2 is on pin 16 on d32
our device connects to speaker but no sound
then i press our test button on our config page our light works but no sound
then i press the reset button on the d 32 the d32 disconnects and reconnects to the speaker
then i press our test button on our config page and it works both the light and sound play 

so it is something to do when i press reset on the d32 that fixes the code issue?
  i attached it here
[naten2.txt](https://github.com/espressif/arduino-esp32/files/3577661/naten2.txt)

everslick, 2019-09-08:
dup of #3173 ?

athleticmistud, 2019-09-08:
This is sub to thst.  3173 is much more detailed with files that support it.  This can merge into thst one

closed 2019-09-08 by athleticmistud