espressif/arduino-esp32 #3174: SD-Card

arnold-b, 2019-09-04:
Hello,

did anyone have a running SD Card. 

all samples and Tips i tried did not work. 
I always get just 
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed

ESP32 Wrover or Wroom Module from  ESPRESSIF
void setup(){
    Serial.begin(115200);
 //   pinMode(4, OUTPUT);

  //  SPI.begin(14, 12, 13, 4);  //void begin(int8_t sck = -1, int8_t miso = -1, int8_t mosi = -1, int8_t ss = -1);
    
    SPI.begin(18, 19, 23);  //void begin(int8_t sck = -1, int8_t miso = -1, int8_t mosi = -1, int8_t ss = -1);
   
    pinMode(23, INPUT_PULLUP);
// No documents found about the frequence setting
    //Testted CS Ports 2,4,5
    if(!SD.begin(5)){//4,SPI)){
        Serial.println("Card Mount Failed");

beegee-tokyo, 2019-09-04:
I have a SD card working, I am using following code:
```cpp
	if (!SD.begin(5))
	{
		log_e("Card Mount Failed");
		// return;
	}

	uint8_t cardType = SD.cardType();

	if (cardType == CARD_NONE)
	{
		log_e("No SD card attached");
		// return;
	}

	/// \todo remove that stuff for final version
	String logMsg = "SD Card Type: ";
	if (cardType == CARD_MMC)
	{
		log_v("%sMMC", logMsg.c_str());
	}
	else if (cardType == CARD_SD)
	{
		log_v("%sSDSC", logMsg.c_str());
	}
	else if (cardType == CARD_SDHC)
	{
		log_v("%sSDHC", logMsg.c_str());
	}
	else
	{
		log_v("%sUNKNOWN", logMsg.c_str());
	}

	uint64_t cardSize = SD.cardSize() / (1024 * 1024);
	log_v("SD Card Size: %lluMB\n", cardSize);
```

But I had a lot of problems to get it running. 
- Not all SD cards work, but I found no pattern to say which card types work.
- Cards work only after formatting on Windows in FAT32 format
- Cards fail frequently because of bad contact from the cards contacts to the card adapters contacts. I have to take them out, clean the contacts on the SD card and then they work again.

Looking for a better solution, but found no replacement so far.

arnold-b, 2019-09-05:
Hello, 
thank you for reply.
I don´t know why, but suddenly it works. 
Not very reliable. The Change is around 75% that it works after reboot

beegee-tokyo, 2019-09-05:
@arnold-b 
What SD card adapter are you using? My problems are definitely linked to a cheap SD card adapter I got from a Chinese supplier. I changed just now to an Adafruit Adalogger board and that works better.

arnold-b, 2019-09-05:
Cust now I´m using no adapter. I´m only using te card slot from china. But this slot works perfect on raspberry

arnold-b, 2019-09-05:
Now my result is diff. After reset the adapter don´t work. If i put out the card and put it in again. Then the Card works.

arnold-b, 2019-09-05:
It works now relaiable. I didn´t use any resitor only one 100nf direct on the 3v3 and Gnd pin

closed 2019-11-18 by stale[bot]