espressif/arduino-esp32 #3176: Crash when coming out of deep sleep

atsage, 2019-09-04:
### Hardware:
Board:							Lolin D32
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    unsure
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I get this crash randomly when coming out of deep sleep, maybe 1 in 100 times? If the ESP would simply reboot it would not be an issue for me but it just completely locks up and needs to be manually reset. Any help would be appreciated.

### Debug Messages:
(Trying to recreate with core debug level set to debug now)

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401ab764: bad00bad bad00bad bad00bad
0x401ab764: gatt_remove_an_item_from_list at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761

Guru Meditation Error: Core  1 panic'ed (IlleGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (?t?X? )
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (

atsage, 2019-09-05:
I can't seem to reproduce this issue with log level set to debug.  I changed back to log level none and got the error pretty much right away, but a different file

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
[... 9 lines cut ...]

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
[... cut ...]
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401aaa9c: bad00bad bad00bad bad00bad
0x401aaa9c: __floatsidf at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:2167

Guru Meditation Error: CGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (,?t?X? )
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (
```

atanisoft, 2019-09-05:
@atsage we can't help you without some code to review. The fact that switching log level to DEBUG fixes it indicates that there is possibly some sort of timing issue as the DEBUG logs will increase delays as various services start back up after returning from sleep.

closed 2019-10-24 by atsage