espressif/arduino-esp32 #3178: Guru Meditation Error trying to send TCP packet while in interrupt handler

sixtyfive, 2019-09-05:
Over at https://github.com/plapointe6/EspMQTTClient/issues/20, we're wondering about a Guru Meditation Error occurs when trying to send a TCP packet while inside an interrupt handler. It's working fine on ESP8266, so perhaps there's a way to make it work on ESP32 by changing something in the Arduino Core?

sixtyfive, 2019-11-04:
Well, I would still be interested in any kind of comment, dear bot.

me-no-dev, 2019-11-04:
you can not do that from interrupt because it involves things that are stopped. You have options to message a task and run whatever you want in it. Look into Queues if you need to pass data or EventBits if you just need to pass a flag like "send a packet now"

closed 2019-11-04 by me-no-dev