espressif/arduino-esp32 #3179: I need the libesp32.a

Nathan-ma, 2019-09-05:
Hello guys. 
recently a fix on the bt lib responsible for making a few devices unable to connect with classic BT was fixed and implemented into v3.2 of esp-idf.
I really need to implement this fix into my arduino code. since it is a esp-idf fix on a supported branch i've been trying, with no success, to create a libesp32.a file to replace in the arduino-esp32 version I have on my arduino IDE.

Here is what I did:

- I got the gif repository cloned with the v3.2 branch, check it out into the proper commit and with the fix I need.
- Installed arduino-esp32 as a component into my projects file (the master one I believe).
- idf.py build on the project folder ( I used the example a2dp_source since my code is based on this example code).
- got into the build folder to get the libesp32.a files and replace on the Arduino IDE, no success.

**Important note:
 - I did enable or disable a few things on the menuconfig in order to make the build compile:
  -- Enabled exception on Cpp
  -- disable HTTPSecure.

What am I doing wrong? how this process should be done? is dere a commit specific to the version 1.0.2 or v3.2? is there a way to compile everything I need without having to enable or disable things on the menuconfig?

Thank you very much

atanisoft, 2019-09-05:
You can use https://github.com/espressif/esp32-arduino-lib-builder to build the appropriate binaries for you in an automated fashion.

Nathan-ma, 2019-09-05:
seems interesting, I will definitely check out.

Nathan-ma, 2019-09-07:
so @atanisoft , am I correct on assuming the link you send me was to convert sketchs (.ino) into the appropriated binaries?

lbernstone, 2019-09-07:
The script installs needed pacakge, downloads esp-idf, and compiles it.  This will create the libraries (including libbt.a) that you can then place into <esp32>/tools/sdk/lib for arduino-esp32 to use.

Nathan-ma, 2019-09-07:
@lbernstone Oh thats all I need? so I dont need to install arduino-esp32 as a Component on the esp-idf project folder? that's awsome Thank you lberstone!!!

lbernstone, 2019-09-07:
Not me.  Thank @me-no-dev

closed 2019-11-21 by stale[bot]

sonolive, 2022-02-07:
hello,
i have done this, everything is ok ... except the libesp32.a that is not recompiled ...

any way what i am looking is : 
in which lib the "SPI RAM config" from menuconfig is hard coded ?
i need a new config with : 
CONFIG_SPIRAM_USE_MEMMAP = 1

but as default it is on
CONFIG_SPIRAM_USE_MALLOC=y

many thanx in advance