espressif/arduino-esp32 #3180: Can't send single byte over BluetoothSerial using serialBT.write

tabroughton, 2019-09-05:
I'm trying to send a single character over bluetooth using SerialBluetooth from Featherboard ESP32.  

The Character isn't sending it is getting queued, how do i make it send.

Example below shows a char being written to Serial and SerialBT - the char is printed in the Serial but not the SerialBT.  Is this a bug or am I doing something wrong?

```
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); 
  Serial.println("Sending single char every 3 seconds:");
}

void loop() {
  Serial.write(45);
  SerialBT.write(45);
  delay(3000);
}
```

beegee-tokyo, 2019-09-05:
Did you try
```
SerialBT.flush();
```

tabroughton, 2019-09-05:
Yes I did @beegee-tokyo, that is making no difference (I thought it would too).  I added it below `Serial.write(45);`

It's worth me mentioning if I run the SerialBT example I can send data from Serial.read() just fine (e.g. `SerialBT.write(Serial.read())`)

beegee-tokyo, 2019-09-05:
Try
```
char single = 45.;
SerialBT.write(&single,1);
```

tabroughton, 2019-09-05:
I get compile error

```
invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
```

atanisoft, 2019-09-05:
try ``` SerialBT.write((uint8_t)45); ``` ?

beegee-tokyo, 2019-09-05:
Forgot the type change 
```
char single = 45;
SerialBT.write((uint8_t*)&single,1);
```

tabroughton, 2019-09-05:
neither of those give the compile error but none of the data is coming through to the bluetooth terminal (on my phone)...  

When I next example the char 45 which is a hyphen (-) is appended to each character so if I type  "foo" into the serial monitor then on my bluetooth terminal I output "-f-0-0-"

The thing I notice with this is the very last "-" meaning there is a serial byte read at the end of the string I typed, would this be the enter key or the termination of the serial buffer and if so is this needed to be passed to ```SerialBT.write``` in order for the queued data to send?

```
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("Sending single char every 3 seconds:");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(45);
    SerialBT.write(Serial.read());
  }
}
```

atanisoft, 2019-09-05:
It is possible that ```Serial.read()`` is returning a null character since it is a binary stream and not guaranteed to be text data.

tabroughton, 2019-09-05:
I have this working now

```
SerialBT.write(45); // "-"
SerialBT.write(13); // carriage return
```
The data now sends or rather it now displays in the Bluetooth terminal app I'm using on my android phone.  It may be that the data was always sending but required a carriage return (I'm assuming this or another similar is the extra byte on my Serial.read())

Question is now, is it the library that needs the carriage return or the android app I'm using.  I would test with other apps if anyone can recommend one.

atanisoft, 2019-09-05:
Likely it is a problem on the android app side in that it is not displaying until it sees a NL/CR. Likely there is a config option to display the raw data as it comes in rather than buffer it.

tabroughton, 2019-09-05:
yes, you are right.  I have tried this now on another terminal app and the original code works fine.  How annoying and I apologies - note that sending a Serial.read() to the terminal app worked fine so this is what threw me (I assumed the app was ok).  This app was highly rated on play store (Arduino Bluetooth Controller) and I've left feedback letting others know if they care to read (there are no configs either).

Thanks for having the conversation with me, it meant I carried on debugging/testing and what could have been an annoyance over a number of days turned out to be just an hour or so.  so thank you.

closed 2019-09-05 by tabroughton